FreeCAD/FreeCAD #21269: Sketcher: Constraint text cutted off in 1.1dev

AndersWestman, 2025-05-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I assume this should be already known, but I can't find any reported issue for it.

The problem: 
Constraints texts are often cutted off in Sketcher.

How to reproduce: 
1. Open a sketch
2. Draw something with several constraints e.g. a hexagon
3. Some texts are cutted off and not shown correct 

I don't know exactly when this issue was first seen, but for me is was somewhere between 40421 and 41214.
1.1.0dev.40421 works
1.1.0dev.41214 and later don't work

[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda
Build date: 2025/04/09 18:28:54
Build type: Release
Branch: main
Hash: de124a72b56515a34f8ed627a768b54695698fc8
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Swedish/Sweden (sv_SE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.25/1.25
Installed mods: 
  * Curves 0.6.40
  * fasteners 0.5.29
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kadet1090, 2025-05-13:
I suspect that this PR may be a culprit: https://github.com/FreeCAD/FreeCAD/pull/21098 I'll check later this week.

Syres916, 2025-05-13:
Certainly looks like scaling is relevant here, using scaling at 100%:

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41671 (Git)
Build date: 2025/05/12 16:21:14
Build type: Release
Branch: main
Hash: a997c1ef16a0a4657e0edf61fb15d01058832227
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9222/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * Assembly4 0.50.18
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

gives:

[image]

and changing to 200%:

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41671 (Git)
Build date: 2025/05/12 16:21:14
Build type: Release
Branch: main
Hash: a997c1ef16a0a4657e0edf61fb15d01058832227
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/47.9969/2
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * Assembly4 0.50.18
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

gives:

[image]

Syres916, 2025-05-13:
> Where do I find this scaling? It's nothing that I have changed as I know.

I'm not a Windows user but doing a quick search, see https://support.microsoft.com/en-gb/windows/change-your-screen-resolution-and-layout-in-windows-5effefe3-2eac-e306-0b5d-2073b765876b specifically the section _Change the size of text, apps, and other items_ on my Linux distro I only have large steps to test but I would imagine you have 100%, 125%, 150% and 200% so you need to put it back to 100% to see if that is a workaround while a proper fix is implemented in FreeCAD's code.

AndersWestman, 2025-05-14:
@luzpaz Regarding the changed title, for me on Windows 11 scaling the screen does not affect the issue. The text is cut off regardless of scaling the screen or changing the font size.

Beep6581, 2025-07-16:
Same problem in Linux/KDE, the bottom halves of constraint labels are cut off:

[image]

This problem appeared in the last few months, was fine before.

Preferences > Sketcher > Display > Sketch Editing > Font Size changes the text size but half is always chopped off.

maxwxyz, 2025-08-19:
@PaddleStroke is this the same issue as https://github.com/FreeCAD/FreeCAD/issues/23128 and fixed by https://github.com/FreeCAD/FreeCAD/pull/23132 ?

maxwxyz, 2025-08-20:
@PaddleStroke seems not fully fixed

Beep6581, 2025-10-07:
@maxwxyz cannot reproduce in 3a8de56ee5cfda0c568f065c208cd888c6f9b05d

Beep6581, 2025-10-08:
@AndersWestman in my comment yesterday I tested the latest commit on master at the time, as per the provided commit hash: 3a8de56ee5cfda0c568f065c208cd888c6f9b05d That is the correct way to identify what you're testing. You tested 68f083a6d169ec523fa9d46b7cc40addb1e4ffbe which was the very next commit, so we essentially tested the same thing. So the issue status is that things seem to have improved but are not entirely solved.

luzpaz, 2025-12-16:
@AndersWestman please **paste** the full [About](http://wiki.freecad.org/About) info (instead of screenshotting it). This is because we can ascertain it quicker instead of needing to click on the screenshot. TIA

Also can you test on 1.1RC1 ?

AndersWestman, 2025-12-16:
I tested the latest 1.2.0dev, but it looks very similar.

```
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.2.0dev.20251209 (Git shallow)
Build date: 2025/12/09 13:53:17
Build type: Release
Branch: grafted,grafted
Hash: 5ad6a1ba58c665c23f1d217c23e74379d8c1a5a4
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Swedish/Sweden (sv_SE)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.25/1.25
Installed mods: 
  * Curves 0.6.40
  * fasteners 0.5.29
```

[image]

closed 2026-09-08 by maxwxyz