FreeCAD/FreeCAD #21272: Core: QAbstractItemView::commitData error when changing font file of Draft ShapeString

Roy-043, 2025-05-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When changing the font file of a Draft ShapeString in the Property editor there is a strange warning:

> QAbstractItemView::commitData called with an editor that does not belong to this view

The attache file was created on Ubuntu. Depending on your OS you may have to recreate it.

[Draft_ShapeString_DejaVuSans-Bold.FCStd.zip](https://github.com/user-attachments/files/20159593/Draft_ShapeString_DejaVuSans-Bold.FCStd.zip)

Steps:
1. Download the attached file.
2. Remove the .zip extension.
3. Open the file.
4. In the property editor remove `-Bold` from the Font File property of the ShapeString.
5. Result: the warning quoted above is displayed. But the updated property is processed correctly.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git)
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-17:
Confirmed on   
```
OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git) AppImage
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.065/1
```

Roy-043, 2025-11-26:
Still relevant.
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.2.0dev.20251118 (Git shallow)
Build date: 2025/11/18 21:54:51
Build type: Release
Branch: main
Hash: be7d41dea75f7f2966d7b431d24f7060c6907339
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.4, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

closed 2026-10-20 by chennes