FreeCAD/FreeCAD #21274: BIM: Strings from BIM/nativeifc/ifc_viewproviders.py are not offered for translation

Roy-043, 2025-05-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

AFAICT the translation strings from .../Mod/BIM/nativeifc/ifc_viewproviders.py are not in .../Mod/BIM/Resources/translations/Arch.ts. This may be due to the presence of .../Mod/BIM/ifc_viewproviders.py.

Forum post:
https://forum.freecad.org/viewtopic.php?p=826467#p826441

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git)
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-05-14:
AFAIK *.ts files are created semi-automatically. So this is not a case of manually editing the mentioned *.ts file. @yorikvanhavre handles this I think.

yorikvanhavre, 2025-05-15:
Indeed this file seems to not have been crawled for strings. I'll have a look at it