FreeCAD/FreeCAD #21297: Draft: move Arch/BIM params out of Draft module

marcuspollio, 2025-05-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Draft `params.py` has many parameters related to the BIM (formerly Arch) module:

https://github.com/FreeCAD/FreeCAD/blob/1de7024a4308828fb1f47f62e2c1a2822e3610bc/src/Mod/Draft/draftutils/params.py#L501-L574

In order to help achieve https://github.com/FreeCAD/FreeCAD/issues/21154, maybe it would be sensible to move these parameters out of the Draft module into the BIM one.

@Roy-043 @yorikvanhavre any thoughts, and how to best proceed?

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS (pop:GNOME/pop/xcb)
Architecture: x86_64
Version: 1.1.0dev.41671 (Git)
Build date: 2025/05/12 16:21:14
Build type: Release
Branch: main
Hash: a997c1ef16a0a4657e0edf61fb15d01058832227
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.2
shiboken 5.15.2, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.7.9, OCC 7.8.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-05-13:
Related issue: https://github.com/FreeCAD/FreeCAD/issues/21298

yorikvanhavre, 2025-05-13:
I agree. I believe only in rare places is that function used in BIM. It should indeed be replaced with direct calls to Getters/Setters, so we make both modules more independent

Roy-043, 2025-05-13:
This needs some more thought IMO. The code in params.py was conceived to address the issue of inconsistent default values. That is a real problem that FreeCAD still suffers from.

I am not sold on the params.py code. But switching back to PARAMS.Get* without an alternative solution to maintain consistency would not be a good idea I think.

The argument that Draft and BIM should be more independent is a strange one. BIM very heavily depends on Draft and many Draft preferences are relevant for BIM.

I think @chennes has discussed the issue of consistent defaults for preferences in the past. But I can't find that ATM.

yorikvanhavre, 2025-05-14:
> but not on Draft to depend on BIM.

Exactly. The idea is to be able to build FreeCAD with Draft enabled but BIM disabled.

This is indeed not really a hard dependency, but params.py should not suppose the BIM ui files will always be there. I haven't checked if it's the case...

Roy-043, 2025-05-14:
If the UI files from the BIM WB are missing there are some messages in the Report view. But other than that there are no major issues with params.py AFAICT. There are some other dependecies that are perhaps more critical (Draft_Clone and importDXF.py).

marcuspollio, 2025-05-14:
Indeed it's probably not major, in fact I just saw the issue with the [Wall offset PR](https://github.com/FreeCAD/FreeCAD/pull/21042) as it adds this [new param](https://github.com/FreeCAD/FreeCAD/blob/87768e4c79b447ed19c0c8ab850afcd86279c7e7/src/Mod/Draft/draftutils/params.py#L563).

As you wrote, [`make_clone.py`](https://github.com/FreeCAD/FreeCAD/blob/9d6f1ad37c1d97bcf669272c3124ce6045691b5b/src/Mod/Draft/draftmake/make_clone.py#L73-L105), `importDXF.py` and [`shape2dview.py`](https://github.com/FreeCAD/FreeCAD/blob/9d6f1ad37c1d97bcf669272c3124ce6045691b5b/src/Mod/Draft/draftobjects/shape2dview.py#L233-L380) have a few `Arch` imports, [`test_svg.py`](https://github.com/FreeCAD/FreeCAD/blob/9d6f1ad37c1d97bcf669272c3124ce6045691b5b/src/Mod/Draft/drafttests/test_svg.py#L82-L101) has a minor one too.

Should we reference them all in https://github.com/FreeCAD/FreeCAD/issues/21154?

pieterhijma, 2025-05-14:
> Should we reference them all in [#21154](https://github.com/FreeCAD/FreeCAD/issues/21154)?

No, not necessary. I'm aware and I'm going to remove BIM as a dependency of Draft. It's runtime checks anyway, so I don't expect any problems.

yorikvanhavre, 2025-05-15:
I guess it's mostly a matter of trying and see what breaks :)