FreeCAD/FreeCAD #21299: Sketcher change value press enter using ketboard  in  textbox crash

symdeb, 2025-05-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Whenever changing value of constraint using  textbox then confirm press enter, it crashes. Flatpack,, Appimage, version 0.20, 0.21, 1.0 all of them, using mouse click "OK" no crash, Ubuntu 24.10. 

flatpak run org.freecad.FreeCAD
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Qt: Session management error: Could not open network socket
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x3ee80) [0x7efe9f2fae80]
#1  0x7efea29fbb5b in Gui::GestureNavigationStyle::AwaitingMoveState::react(Gui::GestureNavigationStyle::Event const&) from /app/freecad/lib/libFreeCADGui.so+0x18b
#2  0x7efea29fc2a1 in boost::statechart::simple_state<Gui::GestureNavigationStyle::AwaitingMoveState, Gui::GestureNavigationStyle::NaviMachine, boost::mpl::list<mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na>, (boost::statechart::history_mode)0>::react_impl(boost::statechart::event_base const&, void const*) from /app/freecad/lib/libFreeCADGui.so+0x21
#3  0x7efea29f4518 in Gui::GestureNavigationStyle::processSoEvent(SoEvent const*) from /app/freecad/lib/libFreeCADGui.so+0x2b8
#4  0x7efea29e89e9 in Gui::NavigationStyle::processEvent(SoEvent const*) from /app/freecad/lib/libFreeCADGui.so+0x69
#5  0x7efea2a1b70f in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /app/freecad/lib/libFreeCADGui.so+0x6f
#6  0x7efea29cb07d in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x14d
#7  0x7efe9f8d6b9e in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x8e
#8  0x7efea08763c8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x78
#9  0x7efea087fd74 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0xb44
#10  0x7efea25c31b8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0xd8
#11  0x7efe9f8d6e40 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x140
#12  0x7efea087e733 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x1f3
#13  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x210bd7) [0x7efea08e8bd7]
#14  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x213aa0) [0x7efea08ebaa0]
#15  0x7efea08763d8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x88
#16  0x7efea25c31b8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0xd8
#17  0x7efe9f8d6e40 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x140
#18  0x7efe9ff950ff in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6+0x82f
#19  0x7efe9fff62dc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6+0xac
#20  /usr/lib/x86_64-linux-gnu/libQt6XcbQpa.so.6(+0x5feb6) [0x7efe99b15eb6]
#21  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x239) [0x7efe9db48e59]
#22  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xbe2b8) [0x7efe9dba62b8]
#23  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7efe9db46513]
#24  0x7efe9fbc891f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x6f
#25  0x7efe9f8e3a4b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x26b
#26  0x7efe9f8dfc4d in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x9d
#27  0x7efea24e8387 in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0xd17
#28  FreeCAD(+0x9607) [0x55ec91929607]
#29  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x2808a) [0x7efe9f2e408a]
#30  /usr/lib/x86_64-linux-gnu/libc.so.6(__l

### Full version info

```shell
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-13:
Can you confirm on weekly build ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

xtemp09, 2025-11-23:
@symdeb, can this issue be closed?