FreeCAD/FreeCAD #21301: Undo of creating a PartDesign SubShapeBinder crashes

Rahix, 2025-05-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using the `PartDesign_SubShapeBinder` command to create a shape-binder in the current active body and then trying to undo this operation leads to the following error message.  This seems to happen in any constellation I have tried, it doesn't seem to matter which object is the shape-binder is created for.

```text
<Exception> invalid container
<App> Transactions.cpp(186): Exception on undo 'Create SubShapeBinder':invalid container
```

I have frequently encountered crashes when interacting with the document after this error.

One such crash can be triggered with the following steps:

1. New document
2. New PartDesign body
3. Create a new sketch, leave it empty
4. Select sketch, them create a SubShapeBinder for it
5. Undo the last operation, error message appears
6. Enter the sketch again by double-clicking
7. Close the sketch
8. FreeCAD crashes

<details>

<summary>Click to expand crash log</summary>

```text
<Exception> invalid container
<App> Transactions.cpp(186): Exception on undo 'Create SubShapeBinder':invalid container
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x3ee80) [0x780e56a51e80]
#1  /app/freecad/lib/libFreeCADApp.so(+0x2b7733) [0x780e596b7733]
#2  0x780e596c4ae7 in App::Document::getDependencyList(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, int) from /app/freecad/lib/libFreeCADApp.so+0x167
#3  0x780e596c7c49 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /app/freecad/lib/libFreeCADApp.so+0x419
#4  0x780e597a017f in App::DocumentPy::recompute(_object*) from /app/freecad/lib/libFreeCADApp.so+0x2ef
#5  0x780e597a02a8 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /app/freecad/lib/libFreeCADApp.so+0x28
#6  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c3116) [0x780e58bc3116]
#7  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x97) [0x780e58bc2f17]
#8  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x2db6) [0x780e58b06196]
#9  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2176) [0x780e58cd2176]
#10  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xaa) [0x780e58cd222a]
#11  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2b24) [0x780e58cd2b24]
#12  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2c1b) [0x780e58cd2c1b]
#13  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x74) [0x780e58cd2ed4]
#14  0x780e5916e92e in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /app/freecad/lib/libFreeCADBase.so+0x6e
#15  0x780e5a3902aa in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /app/freecad/lib/libFreeCADGui.so+0x9a
#16  0x780e5a390456 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /app/freecad/lib/libFreeCADGui.so+0xf6
#17  0x780dd08d55a3 in SketcherGui::TaskDlgEditSketch::reject() from /app/freecad/lib/SketcherGui.so+0x153
#18  0x780e5a721c29 in Gui::TaskView::TaskView::reject() from /app/freecad/lib/libFreeCADGui.so+0x79
#19  /usr/lib/x86_64-linux-gnu/libQt6Core.so.6(+0x1e55fe) [0x780e571e55fe]
#20  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x399608) [0x780e58599608]
#21  /usr/lib/x86_64-linux-gnu/libQt6Core.so.6(+0x1e55fe) [0x780e571e55fe]
#22  0x780e584c9f84 in QAbstractButton::clicked(bool) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x44
#23  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x2ca20f) [0x780e584ca20f]
#24  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x2cb910) [0x780e584cb910]
#25  0x780e584cbb18 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0xf8
#26  0x780e583fd7c8 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x298
#27  0x780e5839e3d8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x88
#28  0x780e583a7f43 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0xd13
#29  0x780e5a3381b8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0xd8
#30  0x780e57182e40 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x140
#31  0x780e583a6733 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x1f3
#32  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x210bd7) [0x780e58410bd7]
#33  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x213aa0) [0x780e58413aa0]
#34  0x780e5839e3d8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x88
#35  0x780e5a3381b8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0xd8
#36  0x780e57182e40 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x140
#37  0x780e579e50ff in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6+0x82f
#38  0x780e57a462dc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6+0xac
#39  /usr/lib/x86_64-linux-gnu/libQt6XcbQpa.so.6(+0x5feb6) [0x780e51799eb6]
#40  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x239) [0x780e557cee59]
#41  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xbe2b8) [0x780e5582c2b8]
#42  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x780e557cc513]
#43  0x780e5747491f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x6f
#44  0x780e5718fa4b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x26b
#45  0x780e5718bc4d in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x9d
#46  0x780e5a25d387 in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0xd17
#47  FreeCAD(+0x9607) [0x55d0ca085607]
#48  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x2808a) [0x780e56a3b08a]
#49  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x780e56a3b14b]
#50  FreeCAD(+0x9955) [0x55d0ca085955]
```

</details>

### Full version info

```shell
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: 
  * Assembly4 0.50.15
  * Curves 0.6.51
  * ExplodedAssembly
  * FreeCAD-library
  * FreeCAD-render
  * ThreadProfile 1.92.0
  * fasteners 0.5.33
  * fcgear 1.0.0
  * kicadStepUpMod 11.4.3
  * sheetmetal 0.5.10
  * Defeaturing 1.2.2
  * FusedFilamentDesign
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-13:
@Rahix can you test on 1.1 dev and report back ?  
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

Rahix, 2025-05-13:
There we go, not reproducable using the version below.  I tried all the variants where I was previously able to trigger this, none lead to the error message or a crash any more.  So I guess this was fixed already!

```text
OS: Arch Linux (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41671 (Git) AppImage
Build date: 2025/05/12 16:21:14
Build type: Release
Branch: main
Hash: a997c1ef16a0a4657e0edf61fb15d01058832227
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
```

luzpaz, 2025-05-15:
Fixed in 1.1dev. Closing