FreeCAD/FreeCAD #21306: TechDraw: right clicking on an object clear selection.

PaddleStroke, 2025-05-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Right clicking on a view for example clear selection.

Expected behavior : right clicking on an object does not clear selection.

Currently this has no consequences because the context menu is non-existent. But a context menu would be unusable with this.
For example : select a view, right click to select "lock view" : not possible because the selection is cleared before the menu appears.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-05-18:
There are clearSelection calls scattered throughout TD.  Most are marked "as a convenience".   They should all be reviewed as I don't think they are helpful in most cases.

Pendelf, 2025-05-23:
If you click in a non-selection it resets in both cases. Without a lock, it does not reset if you click inside the selection. With a lock, it resets if you click inside or outside the selection

https://github.com/user-attachments/assets/59fd7b5d-236a-40cd-91df-b60611d11200

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.62
  * fasteners 0.5.38
  * PieMenu 1.12.0
  * Road 2025.2.22
  * sheetmetal 0.7.22
  * Telemetry 1.0.1

PaddleStroke, 2025-11-13:
RMB should not clear selection if the RMB is on the view. Because in this case you cannot have a contextual menu specific to the selected object.