FreeCAD/FreeCAD #21307: OBJ importer dialog is confusing for new users

furgo16, 2025-05-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The OBJ file import dialog prompts the user to select an importer:

[image]

1. There is no context or guidance as to which one the user should choose
2. Although both are OBJ format importers, their name wildly differs and is rather technical

### Full version info

```shell
1.1.0-dev
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-13:
@FreeCAD/cad-working-group @FreeCAD/design-working-group any proposals?

luzpaz, 2025-08-19:
Note:   
* @wwmayer contributed a fix https://github.com/FreeCAD/FreeCAD/pull/23214 for https://github.com/FreeCAD/FreeCAD/issues/19456