FreeCAD/FreeCAD #21309: Delete group recursively does not delete its contents recursively.

leoheck, 2025-05-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Deleting this

[image]

Ending with this.

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41573 (Git)
Build date: 2025/05/05 17:45:01
Build type: Release
Branch: main
Hash: 3189030c11010631e726dbe350e9077dc84109bb
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-05-13:
Related issue : https://github.com/FreeCAD/FreeCAD/issues/19754#issuecomment-2674712443

closed 2025-09-29 by chennes

leoheck, 2025-09-29:
Awesome!