FreeCAD/FreeCAD #21316: [Sketcher] Futher regression from PR20540, creating a polygon such as Hexagon with Building US scheme does not show Sides as 6 but 0.00 then 3.00

Syres916, 2025-05-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

With a new user config files (therefore Standard/Internal system unit schema), start a new Empty file, switch to the Sketcher Wb and change the document unit schema to `Building US` (num=5).

Start a new sketch and choose X-Y plane (it makes no difference)

On the Create a new Polygon icon drop down to Hexagon, the tasks panel opens and in the `Sides` QuantitySpinBox should be the integer `6` instead it shows `0.00` and after one left mouse click the Sides changes to `3.00` making a triangle!!

A Git Bisect arrives at https://github.com/FreeCAD/FreeCAD/commit/1155f0d75281e9b9d70536d9b81dde7caececadd (PR20540)

### Full version info

```shell
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41455 (Git)
Build date: 2025/04/26 22:45:54
Build type: Release
Branch: (no branch, bisect started on main)
Hash: 1155f0d75281e9b9d70536d9b81dde7caececadd
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9222/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * Assembly4 0.50.18
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38

Even with fixes from Pr #21174 and https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Revert_Internal_metres_to_Standard_millimetres the issue is still present
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-13:
@3x380V I'll have a plough through to see if I can determine which part of the commit is responsible.

Syres916, 2025-05-13:
It looks like the line https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Sketcher/Gui/DrawSketchHandlerPolygon.h#L308 is not finding the correct info for `Base::Unit` and I notice that in the old UnitsApi.h there were two entries:

```
    /// generate a value for a quantity with default user preferred system
    static double toDouble(PyObject* args, const Base::Unit& u = Base::Unit());
    /// generate a value for a quantity with default user preferred system
    static Quantity toQuantity(PyObject* args, const Base::Unit& u = Base::Unit());
```

where now there's only one:

`    static double toDouble(PyObject* args, const Base::Unit& u = Base::Unit());`

More investigation required.

3x380V, 2025-05-13:
@Syres916, for now I can only confirm your findings. Just switching to Building US while editing sketch is enough to hit this bug. I'll look at it tomorrow.

Syres916, 2025-05-14:
@3x380V thanks, all tested fine with the updated PR. As far as the use of a Pref Spinbox in this instance, I can only surmise that seemed the easy way to retain the user's last polygon type selection but it's purely guess work.

3x380V, 2025-05-14:
@Syres916, thank you! Now we need someone to review and merge that PR. I would appreciate if you could the test #21147 once rebased against main with regression fixed applied, so we'll avoid finding regression too late.

PR #21174 (2025-05-14): Base: fix Unit scheme management regressions
Fixes: #21172
Fixes: #21209
Fixes: #21245
Fixes: #21316
Fixes: #21317
closed 2025-05-14 by hyarion