FreeCAD/FreeCAD #21317: Core: Revised Building US unit system cannot read its own output

Roy-043, 2025-05-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The old implementation of the Building US unit system had a similar issue. But now, I think since #20540, the situation seems worse.

1. Open a new file.
2. Change the document unit system to Building US.
3. Enter the following in the Python console:
```
from FreeCAD import Units
Units.Quantity(100, Units.Length).UserString
```
4. Result: `'3-7/8″'`. Note that the double quote is actually a different character (probably the double prime symbol: U+2033). Also before the mentioned PR the minus sign would be a plus sign here.
5. Enter this:
```
Units.Quantity(Units.Quantity(100, Units.Length).UserString)
```
6. Result: `0.0005902777777777778 deg`.
7. Enter this (where the double quote is the correct character):
```
Units.Quantity('3-7/8"')
```
8. Result: `53.974999999999994 mm`.

This may be relevant for, or related to, #21316.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git)
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

3x380V, 2025-05-14:
Quantity parsing did not change, but now we have `toFractional` and `toDms`, both attempting to output correctly formatted string. However, parser cannot distinguish between angular and length values and I have no idea how to fix it.

3x380V, 2025-05-14:
Using `'` and `"` in `toFractional` fixes issue described here (already tested that). But do we really want to take this shortcut? Also _US customary_ schema should be modified to use ASCII.

3x380V, 2025-05-14:
@bofdahof, as an user I would really appreciate if you could return to this issue at some point. For now I reverted to using `'` and `"`, but it would be really nice having consistent output and ability to parse it back.

3x380V, 2025-05-14:
@bofdahof, it seems this deserve shinny new `toLegacyFractional` implementation.

hyarion, 2025-05-14:
> @3x380V I see you closed after ' and " , but did that fix @Roy-043 's other concerns? Quick test indicates not:

I've reopened the issue, let me know if it should be closed again

3x380V, 2025-05-15:
Nice. Makes me wonder if it is even possible to contribute AI generated code.
However, as @Roy-043 pointed out, current parser has problem dealing with US scheme:
```
Units.Quantity('-3\' 3" - 3/8"')
-847.725 mm
```
and positive value is unparsable
```
Units.Quantity('3\' 3" + 3/8"')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ValueError: syntax error
```
That said, I like your original attempt to make things consistent, so I'm temped to partially revert previous commit and use the above implementation as `toLegacyFractional`. I also doubt we will be ever able to correctly parse unified output without hints as it is imposible to know whenever 1" is a length measure or an angle measure.

3x380V, 2025-05-15:
Those problems are there from the beginning; quantity is unable to parse back its output. The only true breaking change you did is changing `'` and `"` into U+2032 and U+2032. Otherwise the parser is equally bad at dealing with US fractional lengths. And being equally bad I see no problem keeping your original version also there just for sake of fixing parser at some point. We could have a new temporary schema and instruct people to give it a try. It is hard to know who's setup break without testing that.

3x380V, 2025-05-15:
In the other words you are proposing to change `toFractional` output exactly as it was before and add tests to verify that. Correct?

3x380V, 2025-05-15:
I regret your decision to leave as soon as possible... I implemented you proposal already, however it is very time consuming to switch between branches, so I'd like to finish at least one issue first and prefer that being #21147 given the fact #21353 just popped up, so would be nice to fix it as well in one go. My machine needs over half an hour to build FreeCAD and touching unit header triggers full rebuild...

3x380V, 2025-05-15:
Please note I'm doing very little on my own here, just incorporating review comments.

3x380V, 2025-05-16:
And here we are. Parser crashes on `1' 2" + 1/4"` input:
<details><summary>Debug output</summary>

```
Starting parse
Entering state 0
Stack now 0
Reading a token
Next token is token ONE ()
Shifting token ONE ()
Entering state 2
Stack now 0 2
Reducing stack by rule 8 (line 51):
   $1 = token ONE ()
-> $$ = nterm num ()
Entering state 20
Stack now 0 20
Reading a token
Next token is token UNIT ()
Shifting token UNIT ()
Entering state 1
Stack now 0 20 1
Reducing stack by rule 29 (line 74):
   $1 = token UNIT ()
-> $$ = nterm unit ()
Entering state 50
Stack now 0 20 50
Reading a token
Next token is token NUM ()
Reducing stack by rule 35 (line 81):
   $1 = nterm num ()
   $2 = nterm unit ()
-> $$ = nterm quantity ()
Entering state 22
Stack now 0 22
Next token is token NUM ()
Shifting token NUM ()
Entering state 3
Stack now 0 22 3
Reducing stack by rule 7 (line 50):
   $1 = token NUM ()
-> $$ = nterm num ()
Entering state 54
Stack now 0 22 54
Reading a token
Next token is token UNIT ()
Shifting token UNIT ()
Entering state 1
Stack now 0 22 54 1
Reducing stack by rule 29 (line 74):
   $1 = token UNIT ()
-> $$ = nterm unit ()
Entering state 50
Stack now 0 22 54 50
Reading a token
Next token is token '+' ()
Reducing stack by rule 35 (line 81):
   $1 = nterm num ()
   $2 = nterm unit ()
-> $$ = nterm quantity ()
Entering state 55
Stack now 0 22 55
Next token is token '+' ()
Reducing stack by rule 5 (line 47):
   $1 = nterm quantity ()
   $2 = nterm quantity ()
-> $$ = nterm input ()
Entering state 19
Stack now 0 19
Next token is token '+' ()
unknown file: Failure
Unknown C++ exception thrown in the test body.
```

</details>
This is not a new regression, so I'm temped to just disable that test.

3x380V, 2025-05-16:
@bofdahof, https://github.com/FreeCAD/FreeCAD/pull/21365/commits/0f9087285ebce3c71c4495d7d4e35fcf50b6217e is ready for review.

PR #21174 (2025-05-14): Base: fix Unit scheme management regressions
Fixes: #21172
Fixes: #21209
Fixes: #21245
Fixes: #21316
Fixes: #21317
PR #21365 (2025-05-19): Base: fix more Unit schema management regressions
Fixes: #21317
Fixes: #21353
Fixes: #21373
Fixes: #21394
closed 2025-05-19 by hyarion