FreeCAD/FreeCAD #21330: Hover pop up in the menu hides options and makes them unclickable

hsvfan-jan, 2025-05-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Just a minor nuisance but a bit annoying nonetheless.
When I go into the menu and want to select an option, the hover tip from an option often overlays where I actually want to click with the mouse.

[image]

In this case, if I want to select "Validate sketch" (and know where it is by heart, as it can't be seen here), clicking on the pop-up (to select Validate sketch underneath as it is highlighted in blue) doesn't seem to do anything. Clicking on the symbol or text that isn't covered by the pop-up works.

It doesn't go away after some time and I can only move it somewhere else if I hover over another option that has one of these pop-ups. In this menu, that's only the first three options.

The pop-up should either move or reduce its borders where it is still active. Another option would be to fade it out after some time, if the mouse is too far away from where it was triggered.

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/05/13 09:39:01
Build type: Release
Branch: main
Hash: 7d57569eb493b929e5647bd9b0ad03b33d1acc45
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-16:
@FreeCAD/design-working-group

luzpaz, 2025-05-16:
Can also confirm this on Linux

```
OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git) AppImage
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/149.623/1
Installed mods: 
  * fcVM
  * FreeCAD-Ribbon 1.9.1
  * SearchBar 1.4.2
  * Silk 0.1.8
  * Telemetry 1.0.1
```

hsvfan-jan, 2025-08-25:
I think this isn't fully resolved. At least in current weekly build

```
OS: macOS 15.6.1
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/20 00:00:55
Build type: Release
Branch: main
Hash: 5e4fe3a3c3af46be530969d323368fee8bfcba3b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Germany (en_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

I still get the following issue. Notice that there is enough space and should display on the right. Not visible in the screenshot, but my mouse was in the middle of the word "sketch" on the "Validate sketch" option. Clicking it didn't do anything

[image]

@tiagomscardoso perhaps you want to take another look? 

@maxwxyz could you reopen this? I don't have permission

MisterMakerNL, 2025-08-25:
I do not have the issue
`OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.43049 +11 (Git)
Build date: 2025/08/25 10:06:20
Build type: Release
Branch: Version_specific_settings
Hash: 47cd1f7bfd2d00af1cda7a0cb2d26df097f27d47
Python 3.12.10, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, IfcOpenShell 0.8.1.post1, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/69.8349/2

`

FEA-eng, 2026-03-04:
@hsvfan-jan Can you confirm if it's still relevant? This issue is OS-specific and doesn't seem to occur on Windows.

tiagomiguelcardoso, 2026-03-14:
If the issue is Mac OS specific I cannot fix it

FEA-eng, 2026-09-21:
Seems still relevant.