FreeCAD/FreeCAD #21334: Error in latest build with redundant constraints

xobes, 2025-05-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1) See screenshots A, B, C depicting actions taken:
a) given the file abcd.FCStd
b) add a line (or a circle, or a point) ... anywhere
c) observe redundant constraint:
```
12:17:50  Redundant constraint is not an autoconstraint. No autoconstraints or additional constraints were added. Please report!
```

[abcd.zip](https://github.com/user-attachments/files/20211854/abcd.zip)

[image]

[image]

[image]

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (fvwm3/xcb)
Architecture: x86_64
Version: 1.1.0dev.41698 (Git)
Build date: 2025/05/14 13:05:56
Build type: Unknown
Branch: development
Hash: 9fb2fbf0f7a0bf0d2a63e6bf843ede87453fa317
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/86.4394/1
Installed mods: 
  * MakerWorkbench 1.0.1
  * fasteners 0.5.38
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xobes, 2025-05-14:
Furthermore, I'm unable to copy the sketch or a part containing the sketch

xobes, 2025-05-14:
The constraint that is marked redundant is between two line segments which are each have "Constrain Block" constraints.  When removed, I am able to continue editing.

xobes, 2025-05-14:
Also, in all fairness, I caused the problem in a build from last week.  I haven't confirmed that applying two Constraint Block constraints on adjacent segments can lead to this problem in the latest build.  The two things happened hours apart.

maxwxyz, 2025-05-15:
@AjinkyaDahale FYI

Syres916, 2025-05-15:
@xobes it didn't immediately catch my attention as to the document unit schema you are using. I had to go back to a build from March in order to fix the document. I'm still not convinced the very latest main with some but not all Unit Schema fixes can handle Imperial Civil yet. But attached is the working file (just remove .zip from the filename) using:

[abcd_Syres.FCStd.zip](https://github.com/user-attachments/files/20233658/abcd_Syres.FCStd.zip)

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40492 (Git)
Build type: Release
Branch: (HEAD detached at e97aa7d47e)
Hash: e97aa7d47e5594b81d8a5789a38876414562074e
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, IfcOpenShell 0.8.1.post1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical/physical DPI: 96/93.1131
Installed mods: 
  * FreeCAD-Ribbon 1.8.1.2 (Disabled)
  * btl 0.9.9
  * FreeCAD-themes 2025.1.7
  * Silk 0.1.6
  * sheetmetal 0.7.22
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.30.3
  * ChangeAppFont
  * Quetzal 1.5.6
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Curves 0.6.61
  * QetWireManager
  * Assembly4 0.50.18
  * A2plus 0.4.68
  * ThreadProfile 1.94.0
```

Personally if you want stability use version 1.0.1 just uploaded today https://github.com/FreeCAD/FreeCAD/releases/tag/1.0.1

xobes, 2025-05-15:
> [@xobes](https://github.com/xobes) it didn't immediately catch my attention as to the document unit schema you are using. I had to go back to a build from March in order to fix the document. I'm [...]

OK, well, thank you.
Didn't know there was a release -- been running latest (pulled from master and built) each time i start FreeCAD so that any bugs I find are worth reporting :-)

I'm trying to use the units for "3D printing, Metric small parts & CNC" -- not sure where it was showing you imperial units.

Anyway, thanks for looking.

luzpaz, 2025-05-17:
> it didn't immediately catch my attention as to the document unit schema you are using. I had to go back to a build from March in order to fix the document. I'm still not convinced the very latest main with some but not all Unit Schema fixes can handle Imperial Civil yet.

@Syres916 this is a Units issue then, in your opinion?

Syres916, 2025-05-18:
@luzpaz my gut says no, unfortunately this will need to be bisected to get to the root cause. The same error was generated by importing a DXF file but we understand the why but at present don't know how it was fixed for both DXFs and SVGs. This just highlights that the Unit Tests are not robust enough to catch some rudimentary user workflows.

xobes, 2025-05-18:
I wouldn't go so far as to call it a rudimentary workflow...  I added tons of constraints long before this problem reared its head and strangely it only popped up after a certain number of further constraints and/or lines were added to the drawing. 

Has anyone else recreated this or confirmed that it happens with my document for them? 

It's easy to work around and I've moved on with my work but in terms of solving the problem moving forward, I wouldn't call this rudimentary and it seems very strange.

Syres916, 2025-05-19:
I won't bore anyone with the reason but that time between mid Feb and mid April is probably the most tiresome period to Bisect so it's probably going to be Wednesday before I can do it due to my machine being in use for 13 hours or so in order to complete it.

Syres916, 2025-05-20:
Unfortunately I'm going to have to delay the bisect until the weekend, the Units testing is taking priority and very time consuming.

drcowsley, 2025-06-30:
I wanted a guide circle made up from two arcs.
Using the origin as centre I drew an unconstrained arc (3DoF), then a second arc connecting those two end points. ( 2DoF ! )

As I did it, I half expected a redundancy message as the second coincidence was using up only one degree of freedom.

I only noticed the message when, intending to have another similarly sectioned circle at a smaller radius, I drew my third arc.   I noticed two messages with the same time stamp:

18:09:53  Autoconstraints cause redundancy. Removing them
18:09:53  Redundant constraint is not an autoconstraint. No autoconstraints or additional constraints were added. Please report!

I couldn't reproduce the error messages even though these were the starting guidelines on a brand new sketch.

Playing around with the origin as the common centre:
1 unconstrained arc                                                  3 degrees of freedom  r A B
another unconstrained arc                                        6 DoF   r A B  s C D
Match one pair of ends                                             4 DoF    r A B D   (r=s  B=C)  
Match the other two ends completing the circle      2 DoF - wrong solver message
The radius was already the same.  Should be            3 DoF    r A B      (D = A)

Cheers

DrC

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

AjinkyaDahale, 2025-07-20:
@Syres916 were you able to do the bisect?

drcowsley, 2025-07-21:
You may already know 1.0.1 has the same degrees of freedom issue I noted in
1.0.0

Cheers,

DrC