FreeCAD/FreeCAD #21336: problem with external geometry references starting on weekly-2025.05.06

tolot27, 2025-05-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I've designed my parts with FreeCAD_weekly-builds-41365 and also created a TechDraw. After upgrading to FreeCAD_weekly-2025.05.06 I immediately get the following errors in the log output window after opening the only TechDraw page in the document:

```log
21:56:43  Dimension Dimension002 value 0.012200 is too small for format specifier: %.1f
21:56:43  Dimension Dimension001 value 0.009000 is too small for format specifier: %.1f
21:56:43  Dimension Dimension value 0.012200 is too small for format specifier: %.1f
```

Creating a new dimension produces the following errors and the value of the dimension is 0,00:
```log
21:56:47  Stele_Vierkant#Sketch_SideHoles: Invalid shape name Vertex8
21:56:47  Stele_Vierkant#Sketch_SideHoles: Invalid shape name Vertex10
21:56:47  <Sketch> SketchObject.cpp(9026): External geometry Stele_Vierkant#Sketch002.e4 missing reference: Sketch001.;g27v1;SKT
21:56:47  <Sketch> SketchObject.cpp(9026): External geometry Stele_Vierkant#Sketch002.e12 missing reference: Sketch001.;g35v1;SKT
```

After opening the above mentioned sketch it is no longer fully constrained. Every element depending on an external geometry is not constrained anymore.

Opening the same file with builds 41365 or weekly-2025.04.21 still works without any issue. I tried to narrow it down and tested weekly-2025.04.28 but this build does not start on my Windows machine. There seem to be a missing dll entry point:
[image]

The bug also persists with weekly-2025.05.13.

I've found #19187 but this issue is unrelated to my problem because the problem does not exist in previous weekly builds.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41585 (Git)
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.205/1
Installed mods: 
  * Assembly4 0.50.18
  * BIM021 2021.12.0
  * Importer3D
  * Movie 2025.1.4
  * Render 2024.12.15
  * sheetmetal 0.7.21

It is reproducible in safe mode.
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-16:
Does it work with the latest Windows build from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds (the ones on the main repo are not perfect for now)?
@3x380V might this be related to the units regression or is this something in Sketcher/TD?

3x380V, 2025-05-16:
@tolot27, are you able to provide minimal example, which shows issue? It is hard to guess what's wrong otherwise.

tolot27, 2025-05-16:
> Does it work with the latest Windows build from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds (the ones on the main repo are not perfect for now)?

No, it doesn't work with the recent version: 1.1.0dev.41694
Build date: 2025/05/13 12:46:54
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c

> [@tolot27](https://github.com/tolot27), are you able to provide minimal example, which shows issue? It is hard to guess what's wrong otherwise.

Yes, i know. I'll try to create one.

tolot27, 2025-05-19:
The problem seem to be related to the weekly builds which likely create corrupted files. I've tried to create a MWE just consisting of a origin-symmetrical square with 110x110 mm and pad it to 10 mm. This works well with release_1.0.0 and weekly-builds-41726 but not with weekly-2025.04.21 or weekly-2025.05.06. After leaving the sketch i get random errors and cannot pad the sketch. Closing the file and opening again is not possible because the files are corrupt.
Opening release_1.0.0.FCStd with either weekly-2025.04.21 or weekly-2025.05.06 works but dimension values are too small.

My assumption is that weekly-2025.04.21, weekly-2025.05.06 and weekly-2025.05.13 are corrupt as weekly-2025.04.28 is. These builds should be removed from the release/tag page, IMHO.

[21336.zip](https://github.com/user-attachments/files/20297915/21336.zip)

3x380V, 2025-05-19:
@tolot27, I just tried to open and modify all files you provided with current main at 76ce8ccfd5b52e520c052d2193d7a244e83d4cb6 (that's with #21365 merged), which worked as expected (also checked TechDraw output). I suggest to wait for next weekly build, retest and report back. Thank you.

tolot27, 2025-05-20:
> [@tolot27](https://github.com/tolot27), I just tried to open and modify all files you provided with current main at [76ce8cc](https://github.com/FreeCAD/FreeCAD/commit/76ce8ccfd5b52e520c052d2193d7a244e83d4cb6) (that's with [#21365](https://github.com/FreeCAD/FreeCAD/pull/21365) merged), which worked as expected (also checked TechDraw output). I suggest to wait for next weekly build, retest and report back. Thank you.

Thanks for investigating it. Please can you confirm that you could also open the test files weekly-2025.04.21.FCStd and weekly-2025.05.06.FCStd? I still cannot open these two files with weekly-2025.05.19 and weekly-builds-41789 which were build about 7 h ago and contain the mentioned commit 76ce8cc.
Nevertheless, my important FCStd file is still showing the too small techdraw dimensions. Newly created techdraw dimensions are still too small and did not reflect the sketcher dimensions correctly. Hence, the wrong "scaling" must be included in the FCStd/XML file(s). Where do I have to look into (which file) to correct it?

tolot27, 2025-05-20:
BTW: weekly-2025.05.19 fails to open "release_1.0.0.FCStd" with the following error:
```
11:29:05  Cannot create object 'Page': (DLL load failed while importing TechDraw: Die angegebene Prozedur wurde nicht gefunden.)
```

weekly-builds-41789 can open the file. Hence, the build (processes) of the two repositories are different.

3x380V, 2025-05-20:
Apologies! I had some noisy debug output enabled, so I missed error messages in report window:
```
13:53:53  Fatal Error at file /tmp/weekly-2025.04.21.FCStd, line 110, char 138
13:53:53  Invalid Document.xml: invalid character 0x1B in attribute value 'Error'
13:54:02  Fatal Error at file /tmp/weekly-2025.05.06.FCStd, line 110, char 142
13:54:02  Invalid Document.xml: invalid character 0x2 in attribute value 'Error'
```
The files `release_1.0.0.FCStd`, `weekly-2025.04.21_2nd_trial.FCStd`,  `weekly-builds-41726.FCStd` and `weekly-builds-41726_2.FCStd` are opening correctly and show this drawing
[image]
Is that expected?

tolot27, 2025-05-20:
> Apologies! I had some noisy debug output enabled, so I missed error messages in report window:
>
> ```
> 13:53:53  Fatal Error at file /tmp/weekly-2025.04.21.FCStd, line 110, char 138
> 13:53:53  Invalid [...]

I can confirm this. These to files are MWEs generated by the weekly version mentioned in the file name. It looks like these two weekly builds are corrupt and hence should be removed from the download link.

> 
> The files `release_1.0.0.FCStd`, `weekly-2025.04.21_2nd_trial.FCStd`, `weekly-builds-41726.FCStd` and `weekly-builds-41726_2.FCStd` are opening correctly and show this drawing
Is that expected?

Yes, that is expected. 

I'll try to reduce my main drawing to get a MWE since newly created files do not show this problem anymore.

tolot27, 2025-05-20:
Okay, I found the reason of the problem. Some time ago and with build 41365, I changed the unit system of the document to MKS. This all works well until I open the file with a newer build, even with build 41789. Then the dimension values are incorrectly small (by a factor of 1000). If I just close the TechDraw page, change the document unit system to Standard (mm, kg, s, °) and open the page again, the dimension values will be shown correctly.
It looks like the unit system stored in the document is not interpreted correctly by build 41365 because changing it between Standard and MKS has no effect. In newer versions this works well.

closed 2025-05-20 by tolot27