FreeCAD/FreeCAD #21343: BIM: BIM_Copy option to copy the base object

semhustej, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**

**Problem description**
Many BIM objects (Walls, Slabs, Windows) use base objects as the basis of their shape. When copying BIM objects using `BIM_copy` tool, the BIM object is copied but the base object remains the same as the original one. This makes the `BIM_copy` still somewhat related to the original object. This may be sometimes desirable but sometimes it is not.
It is quite often confusing to new users. Copy is usually associated with creating independent copy, not linked.

See the current behavior in the video:

https://github.com/user-attachments/assets/dfedcdd1-62ed-4a06-ae1a-0cd1674335c2

**Proposed solution**
Offer a `Copy base object` option to the options of the `BIM_Copy` tool. When this option would be enabled when copying BIM_objects, their base object would be copied too and the new BIM_object would be based on the new bae object.
So if I have a Wall001 based on Sketch001 and used `BIM_Copy` with this option the result will be Wall002 based on Sktech002.

- if the `Copy` option is not enabled in the `BIM_Copy` tool, the `Copy base object` should be greyed out (to inform user that it's not active)
- if BIM object is moved during the copy process, the base object should move too using the same vector
- in my opinion `Copy base object` should be enabled by default for the `BIM_Copy` tool, as I think the 'copy' word means there is no link to the original
- the `Copy base object` settings should be preserved during multiple starts of the `BIM_Copy` tool such as other options
- I think this option would also be visible in the `Draft_Move` tool, which has the same code as the `BIM_Copy`. This may confuse users, but I think the added value of this feature for **BIM_Copy** is worth it
- proposed shortcut is `J` as Copy base ob**J**ect, but may be different. O maybe?

Proposed interface of the task panel:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Anything else?

There is a workaround:

- `BIM_copy` the BIM object
- `BIM_copy` the base object
- set the new base object as a base object for the copied BIM object

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-06-04:
That makes sense to me. I for one was also really confused when I copied some objects, left the model for a few weeks, and then modifying the original objects started modifying all the copies.

yorikvanhavre, 2025-07-07:
Indeed I agree it could be interesting in many scenarios.

The easiest would be to add a "copy base" option to the Draft move command, that only appears when in Copy mode...