FreeCAD/FreeCAD #21345: BIM: Change of Transparency of BIM material is not propagated to objects

semhustej, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When `Transparency` value is changed in BIM Material properties, it is not propagated to objects, which currently have the material applied. Color, for instance, is propagated immediately.

Steps to reproduce:

- open sample file

[BIMmaterial-transparency.FCStd.zip](https://github.com/user-attachments/files/20225361/BIMmaterial-transparency.FCStd.zip)

- select the `Material - transparent` 
- change the `Color` to green and `Transparency` to 80
- the wall with the material assigned changes color accordingly
- the wall with the material assigned does not change the transparency - this is an issue

See behavior in the video:

https://github.com/user-attachments/assets/c303455c-d7be-470d-b21c-904a8eabd59d

I can reproduce in FC 1.)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-15:
You do need to consider collaboration on this one, what if you send the file to a customer who also has the same named custom material but it has different settings, should the app not prompt the user before 'ruining' the objects. This was the same discussion had when sketch colours were brought it so we have a property Boolean to retain the specific colour set.

semhustej, 2025-05-15:
> You do need to consider collaboration on this one, what if you send the file to a customer who also has the same named custom material but it has different settings, should the app not prompt the [...]

- Note: this issue is regarding BIM materials, not the materials from the Material WB. These are currently two different material systems.
- I don't think collaboration requirements change the situation.
- I think BIM materials are saved as part of the FCstd file. There is no FreeCAD-wide installation library as with Material WB materials. So the customer would have to import the material of the same name to the file I sent to him for the case you describe to occur.
- The color of the BIM material does not have the same issue as transparency. The behavior is not consistent across material properties. This is not good UX in my opinion.
- I think it is reasonable to expect that when a material is applied to an object, said object will have the properties of the material. This currently does not happen with transparency in this case.

ErtugrulEge, 2025-05-15:
From the user perspective, if I want to change the trasparency property , I want to see a transparent object in 3D view :)

PR #22061 (2025-06-23): BIM: fix transparency propagation of BIM material
Fixes #21345.