FreeCAD/FreeCAD #21346: FEM compiling fails

edi271, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When compiling the latest version (41710), the following error occurs in lines 1491 and 1498 of the file ../Mod/Fem/App/FemPostFilter.cpp:
error: conversion from ‘char**’ to non-scalar type ‘std::vector<std::__cxx11::basic_string<char> >’ requested
The problem was caused by https://github.com/FreeCAD/FreeCAD/commit/c8ccbeb421b8638f34b0bbbbf36d6a5aa826286b

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41710 (Git)
Build date: 2025/05/14 23:02:55
Build type: Unknown
Branch: master
Hash: 280cdd273c602e4f6b094cd27e5fe7d4270bd546
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, boost 1_74, Eigen3 3.4.0, PySide 5.15.2
shiboken 5.15.2, SMESH 7.7.1.0, xerces-c 3.2.3, OCC 7.5.1
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7275/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marioalexis84, 2025-05-15:
@edi271 What version of VTK are you using?

marioalexis84, 2025-05-15:
That bug is an incompatibility issue with the vtk version; you should install vtk9.
I understand that FreeCAD stopped supporting vtk7 a few months ago.
@cheness, am I right?

luzpaz, 2025-05-15:
Mario i think you meant 'chennes'. But I think @ickby was involved with that, right ?

marioalexis84, 2025-05-15:
I meant Chennes because it's a package support issue. But maybe Ickby knows that too.

marioalexis84, 2025-05-15:
Ok. When I wrote that function, I intentionally ignored the bug with vtk7, assuming (wrongly) that we no longer supported that version.
I'll create a pull request with the fix.

luzpaz, 2025-05-17:
@edi271 any chance you can test the fix in #21375 ?

PR #21375 (2025-05-19): Fem: Add support for VTK < 9.1 in calculator filter - fixes #21346

Add support for VTK < 9.1 in calculator filter .
@edi271 

## Issues

#21346
## Before and After Images