FreeCAD/FreeCAD #21347: Building Euro switches to yards and inches

zbynekwinkler, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using the selector in the bottom right corner to change the unit scheme to Building Euro switches the units to yards and inches and not cm, m2, m3. This can be seen directly in the switcher after the change and also in TechDraw dimensions.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git) AppImage
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * MeshRemodel 1.10.10
  * Manipulator 1.5.7
  * Defeaturing 1.2.2
  * ImportNURBS 1.1.0
  * Design456 0.0.1
  * fasteners 0.5.28
  * ArchTextures
  * WebTools 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

zbynekwinkler, 2025-05-15:
[image]

luzpaz, 2025-05-15:
Known issue. It's being addressed but slowly. See https://github.com/FreeCAD/FreeCAD/labels/Topic%3A%20Units

closed 2025-05-15 by luzpaz

zbynekwinkler, 2025-05-15:
I have gone through the 3 pages of open issues under the linked topic, but I have not been able to identify, which one is this issue duplicate of. 

Is there an issue which would - I don't know - like localize the problem more? Is it problem just with the UI or is there some more fundamental problem in the backend somewhere? Where would one start looking while trying to fix it?

In the meantime, is there a workaround?

luzpaz, 2025-05-15:
@maxwxyz 👆

maxwxyz, 2025-05-15:
That's because it was already closed by https://github.com/FreeCAD/FreeCAD/pull/21174
I assume an additional unit system messed up the order in of the dropdown in the status bar. 
Please comment after checking with the weekly next week if the issue is still persistent