FreeCAD/FreeCAD #21353: Core: getValueAs(FreeCAD.Units.Radian) does not work

Roy-043, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

`getValueAs(FreeCAD.Units.Radian)` does not work.

Paste the following in the Python console:
```python
from FreeCAD import Units
val = Units.Quantity(45, Units.Angle)
print(val.getValueAs("rad"))
print(val.getValueAs(FreeCAD.Units.Radian))
val = Units.Quantity(0, Units.Angle)
print(val.getValueAs("rad"))
print(val.getValueAs(FreeCAD.Units.Radian))
```
Result:
```
0.7853981633974483
1.0
0.0
nan
```

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41671 (Git)
Build date: 2025/05/12 16:21:14
Build type: Release
Branch: main
Hash: a997c1ef16a0a4657e0edf61fb15d01058832227
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-05-15:
The same seems to apply to other units:
```python
print(obj.Length)
print(obj.Length.getValueAs(Units.Metre))
```
Result:
```
15200.0 mm
1.0
```

furgo16, 2025-05-15:
This worked in 1.0:

```python
>>> from FreeCAD import Units
>>> val = Units.Quantity(45, Units.Angle)
>>> print(val.getValueAs("rad"))
0.7853981633974483
>>> print(val.getValueAs(Units.Radian))
0.7853981633974483
>>> val = Units.Quantity(0, Units.Angle)
>>> print(val.getValueAs("rad"))
0.0
>>> print(val.getValueAs(Units.Radian))
0.0
```

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * BIM021 2021.12.0
  * Assembly4 0.50.15
```

But no longer in 1.1.0dev:

```python
>>> from FreeCAD import Units
>>> val = Units.Quantity(45, Units.Angle)
>>> print(val.getValueAs("rad"))
0.7853981633974483
>>> print(val.getValueAs(Units.Radian))
1.0
>>> val = Units.Quantity(0, Units.Angle)
>>> print(val.getValueAs("rad"))
0.0
>>> print(val.getValueAs(Units.Radian))
nan
```

Yet this works:

```python
>>> App.Units.schemaTranslate(App.Units.Quantity("1 mm"), App.Units.getSchema())
('1,00 mm', 1.0, 'mm')
```

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41710 (Git) Snap 1453
Build date: 2025/05/14 23:02:55
Build type: Release
Branch: main
Hash: 280cdd273c602e4f6b094cd27e5fe7d4270bd546
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Assembly4 0.50.17
  * Launcher
```

@Syres916 @3x380V is this another manifestation of regression related https://github.com/FreeCAD/FreeCAD/pull/20540 ?

3x380V, 2025-05-16:
This regression is caused by 77f45159638fe8f99871dee34924ba0622d05d6d

3x380V, 2025-05-16:
Please test #21365

PR #21365 (2025-05-19): Base: fix more Unit schema management regressions
Fixes: #21317
Fixes: #21353
Fixes: #21373
Fixes: #21394
closed 2025-05-19 by hyarion