FreeCAD/FreeCAD #21357: Wall tool cannot use sketch with projected geometry

zbynekwinkler, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. create sketch, create rectangle, close sketch
2. create second sketch, use "create external projection geometry" tool to bring the first rectangle into this sketch, close sketch
3. select second sketch, use the Wall tool

```
15:32:09  Wall001 has a null shape
15:32:09  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAKlcaaC/usr/Mod/BIM/ArchWall.py", line 559, in execute
    if self.layersNum:
       ^^^^^^^^^^^^^^
<class 'AttributeError'>: '_Wall' object has no attribute 'layersNum'
15:32:09  Wall001: '_Wall' object has no attribute 'layersNum'
```

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git) AppImage
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * MeshRemodel 1.10.10
  * Manipulator 1.5.7
  * Defeaturing 1.2.2
  * ImportNURBS 1.1.0
  * Design456 0.0.1
  * fasteners 0.5.28
  * ArchTextures
  * WebTools 1.0.0
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

paullee0, 2025-05-15:
Please provide an example model file for ease of understanding and testing.

kaiwas, 2025-05-15:
I confirm the problem.
Test file (remove the zip extension)

[Problem_EXT.FCStd.zip](https://github.com/user-attachments/files/20230289/Problem_EXT.FCStd.zip)

paullee0, 2025-05-15:
I see.  What does PartDesign do with 'External Geometry'?

ErtugrulEge, 2025-05-15:
I can also confirm
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38
```

paullee0, 2025-05-16:
@zbynekwinkler  Could you elaborate your expected workflow.

Arch/BIM objects are not TNP-tolerant aware at the moment.  Not sure if sketch is or how it is supported.  There is feature in external add-on to bring in external geometry from another sketch, but it is not exposed, would like to see peoples' expected workflow and behaviour to see what should be done.

Thanks.

kaiwas, 2025-05-16:
> Is this Toponaming-tolerant now?

No. External geometry still does not take into account changes in the original sketch.
But still, this could be useful in some cases.

[image]

PR #21401 (2025-05-26): [ArchWall] Fix SetLength Error When Base Sketch has External Geometry
fix #21357 error message problem

Github issue and discussion:
https://github.com/FreeCAD/FreeCAD/issues/21357
https://github.com/FreeCAD/FreeCAD/issues/21357#issuecomment-2886010914

ArchWall return error when Base Sketch has external(projected) geometry.

External projected geometry has TNP-problem, until the feature is TNP-tolerant,  support of external(projected) geometry at ArchWall is not encouraged. The reported error message is fixed in the meantime in this PR.

## Issues

## Before and After Images
commit 8ff5c711 (2025-05-26): [ArchWall] Fix SetLength Error When Base Sketch has External Geom

fix #21357 error message problem

Github issue and discussion:
https://github.com/FreeCAD/FreeCAD/issues/21357
https://github.com/FreeCAD/FreeCAD/issues/21357#issuecomment-2886010914

ArchWall return error when Base Sketch has external(projected) geometry.

External projected geometry has TNP-problem, until the feature is TNP-tolerant,  support of external(projected) geometry at ArchWall is not encouraged. The reported error message is fixed in the meantime in this PR.
closed 2025-05-26 by yorikvanhavre