FreeCAD/FreeCAD #21358: Sketcher: Sketch appearance settings outside of edit mode are reset when the file is reopened.

kaiwas, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If you customize the appearance of the sketch (colors), then when you reopen the file, the appearance will be reset. The settings specified in the global parameters are applied.

Video demonstration

https://github.com/user-attachments/assets/39a056e1-0a06-448c-ba93-d50624edb160

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git) AppImage
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master.zip
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * TrackingGroups
  * Curves 0.6.62
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Sketcher

### Anything else?

Related issue

https://github.com/FreeCAD/FreeCAD/issues/21146

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Pendelf, 2025-05-23:
Confirmed, color settings are reset to default after saving and opening

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.62
  * fasteners 0.5.38
  * PieMenu 1.12.0
  * Road 2025.2.22
  * sheetmetal 0.7.22
  * Telemetry 1.0.1

maxwxyz, 2025-07-05:
@kadet1090 is this related to auto-color?

Roy-043, 2025-11-11:
> [@kadet1090](https://github.com/kadet1090) is this related to auto-color?

Yes it is.

Roy-043, 2025-11-11:
@kaiwas I think this can be closed. Do you agree?