FreeCAD/FreeCAD #21364: BIM: Subtracted roof is hidden when the base Wall is copied

semhustej, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Subtracted roof is hidden when the base Wall is copied

Steps to reproduce:

- open sample file (remove the .zip extension)

[FC_roof_remove.FCStd.zip](https://github.com/user-attachments/files/20232403/FC_roof_remove.FCStd.zip)

- select the wall
- start `BIM_Copy` tool and create a copy of the wall
- the copy is create but the roof state is changed to hidden (which the user did not want)

See behavior in the video:

https://github.com/user-attachments/assets/d650ff74-6858-4f0f-92f7-09d578952e98

I am able to reproduce in FreeCAD 1.0

**Desired behavior**
The roof should remain visible after the `BIM_Copy` command is ended.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-06-25:
This issue is related to #17750. The root cause in both cases is the same: setting `self.Type` is done in, or triggered by, `onDocumentRestored` which occurs almost last when a document is opened, or an object copied. Setting it in `loads` solves both issues.

closed 2025-06-30 by chennes

semhustej, 2025-07-26:
Tested with latest weekly, the issue is solved. Thank you.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42721 (Git)
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

commit d8889c3c (2025-06-30): BIM: fix setting of self.Type

Fixes #21364.

`self.Type` should be set in `__init__` and `loads`, and not in `onDocumentRestored`.

Additionally:
fixed mistake in `loads` in ifc_objects.py.