FreeCAD/FreeCAD #21367: FC does not try to recover the design after a crash anymore.

leoheck, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

After crash (chamfer related) freecad is not trying to recover my design.
Instead it reports me this

```
➜ freecad_weekly_apprun 
FreeCAD 1.1.0, Libs: 1.1.0devR41694 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Initializing Assembly4 workbench (0.50.19-dev) ............................. done.
^CError calling Python override of QPushButton::changeEvent(): Traceback (most recent call last):
  File "/home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/Mod/Tux/NavigationIndicatorGui.py", line 45, in changeEvent
    def changeEvent(self, event):

KeyboardInterrupt
```

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-05-15:
The crash msg is this one by the way.

```
➜ freecad_weekly_apprun             
FreeCAD 1.1.0, Libs: 1.1.0devR41694 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Initializing Assembly4 workbench (0.50.19-dev) ............................. done.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45810) [0x74a7f3245810]
#1  0x74a7a222aa94 in Extrema_ExtPC::Perform(gp_Pnt const&) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/./libTKGeomBase.so.7.8+0x7b4
#2  0x74a7a222cc26 in Extrema_ExtPC::Extrema_ExtPC(gp_Pnt const&, Adaptor3d_Curve const&, double) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/./libTKGeomBase.so.7.8+0x276
#3  0x74a7a043f178 in BRepBlend_Walking::CorrectExtremityOnOneRst(int, double, double, double, gp_Pnt const&, double&, double&, gp_Pnt&, double&) const from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0x1f8
#4  0x74a7a0446fe0 in BRepBlend_Walking::InternalPerform(Blend_Function&, Blend_FuncInv&, double) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0x2010
#5  0x74a7a04488ad in BRepBlend_Walking::Perform(Blend_Function&, Blend_FuncInv&, double, double, double, double, double, math_VectorBase<double> const&, double, bool) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0x48d
#6  0x74a7a0360c3a in ChFi3d_Builder::ComputeData(opencascade::handle<ChFiDS_SurfData>&, opencascade::handle<ChFiDS_ElSpine> const&, opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepBlend_Line>&, opencascade::handle<Adaptor3d_Surface> const&, opencascade::handle<Adaptor3d_TopolTool> const&, opencascade::handle<Adaptor3d_Surface> const&, opencascade::handle<Adaptor3d_TopolTool> const&, Blend_Function&, Blend_FuncInv&, double, double, double, double, double&, double&, bool, bool, bool, math_VectorBase<double> const&, int&, int&, bool&, bool&, bool&, bool&, bool, bool) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0x6fa
#7  0x74a7a03b0bf9 in ChFi3d_ChBuilder::PerformSurf(NCollection_Sequence<opencascade::handle<ChFiDS_SurfData> >&, opencascade::handle<ChFiDS_ElSpine> const&, opencascade::handle<ChFiDS_Spine> const&, int, opencascade::handle<BRepAdaptor_Surface> const&, opencascade::handle<Adaptor3d_TopolTool> const&, opencascade::handle<BRepAdaptor_Surface> const&, opencascade::handle<Adaptor3d_TopolTool> const&, double, double, double, double&, double&, bool, bool, bool, bool, bool, math_VectorBase<double> const&, int&, int&) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0x8e9
#8  0x74a7a0348d4e in ChFi3d_Builder::CallPerformSurf(opencascade::handle<ChFiDS_Stripe>&, bool, NCollection_Sequence<opencascade::handle<ChFiDS_SurfData> >&, opencascade::handle<ChFiDS_SurfData>&, opencascade::handle<ChFiDS_ElSpine> const&, opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepAdaptor_Surface> const&, opencascade::handle<BRepAdaptor_Surface> const&, gp_Pnt2d const&, gp_Pnt2d const&, opencascade::handle<Adaptor3d_TopolTool> const&, opencascade::handle<BRepAdaptor_Surface> const&, opencascade::handle<BRepAdaptor_Surface> const&, gp_Pnt2d const&, gp_Pnt2d const&, opencascade::handle<Adaptor3d_TopolTool> const&, double, double, double, double&, double&, bool, bool, bool, bool, bool, math_VectorBase<double>&, int&, int&, opencascade::handle<BRepAdaptor_Surface>&, opencascade::handle<BRepAdaptor_Surface>&) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0x54e
#9  0x74a7a034daff in ChFi3d_Builder::PerformSetOfSurfOnElSpine(opencascade::handle<ChFiDS_ElSpine> const&, opencascade::handle<ChFiDS_Stripe>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, bool) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0x182f
#10  0x74a7a0351c71 in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0xd1
#11  0x74a7a0355faa in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0x6a
#12  0x74a7a031a484 in ChFi3d_Builder::Compute() from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0x4a4
#13  0x74a7a049abb6 in BRepFilletAPI_MakeChamfer::Build(Message_ProgressRange const&) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/././libTKFillet.so.7.8+0x36
#14  0x74a7a2a69a4d in BRepBuilderAPI_MakeShape::Shape() from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/./libTKTopAlgo.so.7.8+0x8d
#15  0x74a7a311d37d in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/./Part.so+0x9d
#16  0x74a7a311d523 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, Part::TopoShape const&, char const*) from /home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/lib/./Part.so+0x83
[... cut ...]
FreeCAD 1.1.0, Libs: 1.1.0devR41694 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Initializing Assembly4 workbench (0.50.19-dev) ............................. done.
^CError calling Python override of QPushButton::changeEvent(): Traceback (most recent call last):
  File "/home/lheck/Applications/FreeCAD-41694/squashfs-root/usr/Mod/Tux/NavigationIndicatorGui.py", line 45, in changeEvent
    def changeEvent(self, event):

KeyboardInterrupt

```

leoheck, 2025-05-15:
Sample file is here: [chamfer_crash.FCStd.zip](https://github.com/user-attachments/files/20232779/chamfer_crash.FCStd.zip)

No matter the line I select on the highlighted edges of the chamfer, it crashes.

[image]

I feel the shape is not well formed since it has this odd vertical line.

[image]

luzpaz, 2025-05-15:
Related to ? 
* https://github.com/FreeCAD/FreeCAD/issues/18044

leoheck, 2025-05-16:
I don't think so @luzpaz.

Here the issue is new on this weekly version or a version very close to this one.

Freecad crashes, and it doesn't offer me to recover the file. It just complains about something in the Report View related to this `QPushButton::changeEvent()`.