FreeCAD/FreeCAD #21369: PartDesign: Chamfer after AdditiveLoft crashes freecad

leoheck, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FC crashing still with chamfers.

Sample file is here: [chamfer_crash.FCStd.zip](https://github.com/user-attachments/files/20232779/chamfer_crash.FCStd.zip)

No matter the line I select on the highlighted edges of the chamfer, it crashes.

[image]

I feel the shape is not well formed since it has this odd vertical line.

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.61
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.0
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.9
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-05-15:
Chamfering with a SubtractivePipe produces some weird artifacts.
[chamfer_crash_2.FCStd.zip](https://github.com/user-attachments/files/20232896/chamfer_crash_2.FCStd.zip)

[image]

leoheck, 2025-05-15:
The issue does not happen if the Loft is `Ruled`

[image]

[image]

leoheck, 2025-05-15:
Maybe releated to https://github.com/FreeCAD/FreeCAD/issues/5642

leoheck, 2025-05-15:
Ah, using a different profile for the Chamfer with the SubtractivePipe I could remove the artifacts.

[image]

maxwxyz, 2025-05-16:
@pieterhijma is this crash OCC related?

PeteHG99, 2025-05-20:
Crashes for me as well on Version: 1.1.0dev.41736 (Git) AppImage

[dump](https://github.com/user-attachments/files/20354087/log.txt)

Fillet works by the way.

pieterhijma, 2025-06-13:
> [@pieterhijma](https://github.com/pieterhijma) is this crash OCC related?

Yes, it is. I've added it to the project.

pieterhijma, 2026-09-30:
Reported in https://github.com/Open-Cascade-SAS/OCCT/issues/1576 with DRAW script [crash-on-chamfer-complex-shape.zip](https://github.com/user-attachments/files/32854420/crash-on-chamfer-complex-shape.zip).