FreeCAD/FreeCAD #21372: Part Design: Pocket fails on

mapandre, 2025-05-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Creating a Pocket from a Sketch containing a single circle fails on a solid which consists of a revolve, a groove and another pocket.

[image]
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * ArchTextures
  * Assembly4 0.50.18
  * Curves 0.6.54
  * Defeaturing 1.2.6
  * ExplodedAssembly
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Marz 0.1.15
  * QuickMeasure 2022.10.28
  * Render 2024.12.15
```

### Subproject(s) affected?

None

### Anything else?

The .zip containes the FCStd file. I named the sketch for the failing pocket "Pockt_fails" and the failing pocket "failing_pocket".

[Pocket_fails.zip](https://github.com/user-attachments/files/20234657/Pocket_fails.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-05-16:
> ### Problem description
> Creating a Pocket from a Sketch containing a single circle fails on a solid which consists of a revolve, a groove and another pocket.

**Problem confirmed as described.**

Version:
_OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda_

Further investigation reveals the problem appears to be caused by seam edge overlapping issue between the pocketing sketch and one of the seam edges of the main body. A slight rotation of the sketch on it's local Z axis results in a successful pocket. 

[image]

maxwxyz, 2025-05-16:
@pieterhijma OCC issue?

FEA-eng, 2025-11-26:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251126 (Git shallow)
Build date: 2025/11/26 00:30:17
Build type: Release
Branch: (HEAD detached at de230cf)
Hash: de230cfaa75433441366ad09a4d5fda828e7bfdb
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```