FreeCAD/FreeCAD #21382: BIM/Core: Add "activate with double click" property to groups

kaiwas, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

At the moment we cannot activate a group by double-clicking in the tree view.
The only option is to select the active group in the panel. In this case, the active container still has priority.
To place an object inside a group, we have to deactivate the container object.

We need to add the ability to activate a group by double-clicking.

Considering that groups do not have placement properties, the working plane for groups can be used from a previously selected container object that has the placement property (for example, a level).

This will allow you to place new objects inside groups without having to select an active group in the panel.

In short, a group should behave the same as a container object (a level or a building), but take the coordinates of the working plane from the container inside which the group is located.

https://github.com/user-attachments/assets/89ce821b-e241-40b2-8e02-9548056fd761

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git) AppImage
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master.zip
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * TrackingGroups
  * Curves 0.6.62
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-05-16:
Agree with @maxwxyz: consistency is very important.

markkomartin, 2025-05-24:
> Draft (and BIM) use a drop-down layers selector that can (there is a pref option for that) be extended to also show groups.
> 
> Maybe we could start using that? For ex, implementing a command, with a keyboard shortcut, that sets the selected object as activated? That would be almost as efficient as double-clicking...
> 
> Another path would be to implement that as an additional edit mode

NEMETSCHEK ALLPLAN has a similar system. 
Each level is organized into various "drawing files" that can be compared to freecad groups.
The active "drawing file" is red.
If I want to see another non-editable  "drawing file" in the background I select the gray icon.
If I want to see another editable  "drawing file" in the background I select the yellow icon.
I worked with Allplan and it was very good.

Watch the following video from minute 2:24 onwards:
https://www.youtube.com/watch?v=l716BGvXmbI&list=PLdOdniujLua2LRa7L-AQ74BGgvssaRH8r&index=4&ab_channel=ALLPLANEnglish

This is another video where you can see the system explained:
https://www.youtube.com/watch?v=ubQhpJcBkcE&ab_channel=AlessioCaggiano

I would be curious to see how Allplan translates these  "drawing file" into IFC .
We could use the same method with Freecad groups.
Unfortunately I don't have a version of Allplan available.