FreeCAD/FreeCAD #21383: Fast switch to English

Mannshoch, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Problem: 
- Many screenshots are taken in the Language of the user. For cases where you use it for your own. This is nice and help full but not if you use it in the FreeCAD community.
- many times you follow a tutorial from a website and need to follow english description and screenshots

In both cases it would be helpfull to have a way to fast switch to FreeCAD in english and back.

Ideas how to solve this:

- help menu entry "temporary switch to english"
- right-click on window title bar "temporary switch to english"
- right-click on toolbar "temporary switch to english"
- Key-combo to fast switch between temporary english and local language

In any case it switch back after closing FreeCAD.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41718 (Git) Snap 1454
Build date: 2025/05/15 21:31:55
Build type: Release
Branch: main
Hash: dc5846808810ac1b748aee9bd636186f3372ed03
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.0, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9608/1
Installed mods: 
  * Movie 2025.1.4
  * ModernUI (Disabled)
  * InventorLoader 1.5.1
  * Render 2024.12.15
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-16:
Maybe this can be integrated into `Restart in safe mode` as we expect every issue to be reported in safe mode 
@hyarion is this possible?

hyarion, 2025-05-16:
@maxwxyz isn't safe mode already in English by default?

maxwxyz, 2025-05-16:
@hyarion no, system locale (at least on Windows)

FEA-eng, 2025-05-18:
I agree that safe mode should be in English by default. When you start it and change the language back to English, some items still remain translated to your local language. And you can't just restart to get rid of that like in the standard mode.

I was also thinking about a quick language change button but for another reason - it would be helpful for translators.

maxwxyz, 2025-05-18:
Also, some languages have issues, e.g. in Korean or Russian you cannot open the preferences, so English in safe mode would be great.

Mannshoch, 2025-05-19:
I'm not sure If the request for English in Safe mode and my request for fast switch are similar enough. It may depends on the cause, why do you need a switch. For my cases safe mode not really work because you also loose your settings while using it.