FreeCAD/FreeCAD #21385: PartDesign/Sketcher feature: allow autocreate a SubShapeBinder when external "object belongs to another body, can't link"

FlachyJoe, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

What about a tool option for Sketcher_Projection and Sketcher_Intersection to allow user to pick a SubShape of another Body with auto-creation of the corresponding SubShapeBinder?

Currently, user who forget to create the binder has to close the sketch editor, pick the subshape, create the binder and return to the sketch. A unique checkbox could save this.

If the user picks multiple subshapes during the sketch edition they could be append to one binder in order not to crowd the tree.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41698 (Git)
Build date: 2025/05/14 13:05:56
Build type: Unknown
Branch: main
Hash: 9fb2fbf0f7a0bf0d2a63e6bf843ede87453fa317
Python 3.13.3, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 85/86.4394/1
Installed mods: 
  * Render 2024.12.15
  * Curves.backup1744720429.3214982 0.6.60 (Disabled)
  * lattice2 1.0.0
  * Lithophane
  * Curves 0.6.61
  * fasteners 0.5.38
  * Assembly4 0.50.18
  * Telemetry 1.0.0
  * Road 2025.2.22
  * FCPDWorkbench 0.4.0
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-16:
@PaddleStroke FYI
IMO a good idea until it is hopefully not necessary anymore. I'm not sure if it needs a checkbox in the tool or in the preferences. Ideally, when there is nothing from the own body where I click, create a binder.

MisterMakerNL, 2025-05-16:
You are entering dangerous TNP area here. My manger would fire you on the spot for having such dangerous idea's.
 Personally I like to live dangerous, so I open for testing this out, but we should definitely not expose this to soon to any normal user..

pierreporte, 2025-05-16:
I agree with this proposal.

@MisterMakerNL You are right. We need a way to break the link between the shape binders and their target, like Catia can isolate linked geometry. For instance extracts can become regular lines, planes, surfaces, etc. I often work like this: create geometry by linking other parts temporarily, then break the link. Though Catia can make them isolated automatically at creation.

FlachyJoe, 2025-05-17:
@MisterMakerNL  I abuse binders and rarely have insurmountable problems. In my mind TNP is not a calamity. The soft is fully usable with it and its mitigation can only improve in the future.

@pierreporte Binder isolation should only append with a user action. If dependencies break automatically, I'll consider it a bug.

pierreporte, 2025-05-17:
@kadet1090 of course. This automatic isolation should be a program setting, so it is the user that decides who decides.

MisterMakerNL, 2025-05-17:
I like all of your idea's :)