FreeCAD/FreeCAD #21389: Draft: Failing test

pieterhijma, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

There is a failing test for Draft:

```
======================================================================
ERROR: test_copy (drafttests.test_modification.DraftModification.test_copy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.13/unittest/case.py", line 58, in testPartExecutor
    yield
  File "/usr/lib/python3.13/unittest/case.py", line 651, in run
    self._callTestMethod(testMethod)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/lib/python3.13/unittest/case.py", line 606, in _callTestMethod
    if method() is not None:
       ~~~~~~^^
  File "/home/pieter/scratch/build-freecad-main/Mod/Draft/drafttests/test_modification.py", line 79, in test_copy
    obj = Draft.move(line, c, copy=True)
  File "/home/pieter/scratch/build-freecad-main/Mod/Draft/draftfunctions/move.py", line 120, in move
    newobj = make_copy.make_copy(obj)
  File "/home/pieter/scratch/build-freecad-main/Mod/Draft/draftmake/make_copy.py", line 73, in make_copy
    newobj = App.ActiveDocument.copyObject(obj)
ValueError: {'sclassname': 'N4Base11PyExceptionE', 'sErrMsg': 'Empty module name', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Empty module name', 'btranslatable': False, 'breported': False}
```

To reproduce, open the Test Framework workbench and run the test "TestDraft".

### Full version info

```shell
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.41717 (Git)
Build date: 2025/05/15 17:23:03
Build type: Unknown
Branch: main
Hash: b54898f05f58e1f9908deb5751e532bc71e8e1ab
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

Draft

### Anything else?

First discovered in https://github.com/FreeCAD/FreeCAD/issues/21154.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-05-16:
> run the test "TestDraft".

How do you do that? I do not see this in a menu.

In my test there are no issues (paste this in the Python console):
```python
import Test, TestDraft
Test.runTestsFromModule(TestDraft)
```

Version:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41671 (Git)
Build date: 2025/05/12 16:21:14
Build type: Release
Branch: main
Hash: a997c1ef16a0a4657e0edf61fb15d01058832227
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

furgo16, 2025-05-16:
Having had to do this recently, and if I understand the question, I hope this helps:

[image]

On the Test Framework workbench, you need to click on the `Self-test` command to select the main test module that each workbench exposes.

Alternatively, I tend to run the tests on the command line:

```shell
./build/bin/FreeCAD --console -t TestDraft
```

Roy-043, 2025-05-17:
Thanks for explaining. Using Test Framework there are no errors either on my system (see info above). So it would seem that this is due to 3rd party libraries.

pieterhijma, 2025-05-18:
Ok thanks, I'll investigate more.

Roy-043, 2025-05-19:
> check that the objects have (hidden) `_ObjectUUID` properties or not?

The objects do not have that property. I do not use a debug build (I do not compile myself).