FreeCAD/FreeCAD #21390: Draft: Remaining dependencies on BIM

pieterhijma, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Compiling FreeCAD without BIM shows remaining dependencies on BIM in the Draft workbench.

Firstly, the following messages are reported on startup:

```
12:54:07  Preferences file :/ui/preferences-arch.ui not found
12:54:07  Preferences file :/ui/preferences-archdefaults.ui not found
12:54:07  Preferences file :/ui/preferences-dae.ui not found
12:54:07  Preferences file :/ui/preferences-ifc.ui not found
12:54:07  Preferences file :/ui/preferences-ifc-export.ui not found
12:54:07  Preferences file :/ui/preferences-sh3d-import.ui not found
```

Then there is a failing test:

```
======================================================================
ERROR: test_workingplaneproxy (drafttests.test_creation.DraftCreation.test_workingplaneproxy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.13/unittest/case.py", line 58, in testPartExecutor
    yield
  File "/usr/lib/python3.13/unittest/case.py", line 651, in run
    self._callTestMethod(testMethod)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/lib/python3.13/unittest/case.py", line 606, in _callTestMethod
    if method() is not None:
       ~~~~~~^^
  File "/home/pieter/scratch/build-freecad-fix-cyclic-build-dep/Mod/Draft/drafttests/test_creation.py", line 353, in test_workingplaneproxy
    obj = Draft.make_workingplaneproxy(placement)
  File "/home/pieter/scratch/build-freecad-fix-cyclic-build-dep/Mod/Draft/draftmake/make_wpproxy.py", line 52, in make_workingplaneproxy
    ViewProviderWorkingPlaneProxy(obj.ViewObject)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/home/pieter/scratch/build-freecad-fix-cyclic-build-dep/Mod/Draft/draftviewproviders/view_wpproxy.py", line 75, in __init__
    vobj.LineColor = params.get_param_arch("ColorHelpers") | 0x000000FF
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
TypeError: unsupported operand type(s) for |: 'NoneType' and 'int'
```

To reproduce, build FreeCAD with CMake flag `BUILD_BIM` set to `OFF`, then start the application, open workbench Test Framework and run the test "TestDraft".

### Full version info

```shell
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.41717 (Git)
Build date: 2025/05/15 17:23:03
Build type: Unknown
Branch: fix-cyclic-build-dep
Hash: b54898f05f58e1f9908deb5751e532bc71e8e1ab
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

Draft

### Anything else?

PR https://github.com/FreeCAD/FreeCAD/pull/21331 removed the erroneous BIM dependency from Draft.

The problem was first noticed in [this comment](https://github.com/FreeCAD/FreeCAD/issues/21154#issuecomment-2881125806). 

[This comment](https://github.com/FreeCAD/FreeCAD/issues/21154#issuecomment-2881281578) from @tritao may help fixing it.

Issue https://github.com/FreeCAD/FreeCAD/issues/21297 is related but is considered a feature. Because of the failing test, the fact that the problems described here are very concrete, I chose to create a separate issue that can be marked as a bug.

https://github.com/FreeCAD/FreeCAD/issues/21389 is a failing test as well, but that is independent from this issue. However, it is also reported as a failing test when running the "TestDraft" test suite. (So, in total two failing tests are reported, but only one is part of this issue).

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-05-16:
> Firstly, the following messages are reported on startup:

This should be fixed with: #21370.

Roy-043, 2025-05-16:
There are several places in the Draft code where an attempt is made to import the Arch module, and if that fails appropriate action is taken. What happens in the params.py code is quite similar. I can understand how this may be disturbing on a principle level, but in practice this is a non-issue IMO.

There are also places where Arch code is referenced without any checks. Let's focus on those instead.

Before investing a lot of time in the code of params.py I think @yorikvanhavre should clarify his position. The code was added during the V1.0 cycle, but it is apparent that Yorik is not a big fan. As I have said in another issue: I am not sold on this solution, but doing away with it without an alternative method to avoid inconsistent defaults seems counter-productive.

PR #21404 (2025-05-17): Draft: use different color for WPProxy to avoid BIM dependency
Fixes #21390.
commit 61e349ef (2025-05-17): Draft: use different color for WPProxy to avoid BIM dependency

Fixes #21390.
closed 2025-05-17 by Roy-043