FreeCAD/FreeCAD #21391: Sketcher: 3-point symmetry constraint: origin should not be special

smilingthax, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using the symmetry constraint with three (regular) points, the result is the same, regardless of the ordering of the first to points ("base line"; the third point becomes the "mid point").

This is not entirely true when the Origin is involved:
* When first selecting a regular point, then the origin, and then a (regular) "mid" point, everything works as expected.
* when first selecting the Origin, then selecting a regular point, there are two cases:
  - In the 'all the points were selected before the tool is applied'-workflow, it works,
  - but, when first activating the tool and then selecting three points, no regular points is allowed to be chosen as third ("mid") point (mouse cursor becomes "forbidden"). This does not make sense.

### Full version info

```shell
OS: macOS 14.6.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-16:
what is the expected outcome? The origin is fixed and cannot be moved.

smilingthax, 2025-05-16:
This has nothing to do with solvers ability to find solutions to `(a+b)/2 = c` where some points are fixed (e.g. `a=0`) – which it is perfectly capable of.

This is a UI issue, where the exact same constraint can be created in one way – by first selecting the points, with the origin first, and then selecting the symmetry constraint tool–, vs. not being able in another way – by first activating the constraint tool, and then choosing the points, with the origin first.

PR #21612 (2025-06-23): Sketcher: 3 Point Symmetry fixed if root is selected first (and simplify and fix the selection logic)
## Issues
fixes #21391

## Summary
The Selection SelVertexOrRoot is ignored if there is a Selection that fits better (SelRoot or SelVertex in this case) See comment on line 985 in CommandConstraints.cpp. So for each SelVertexOrRoot where an option exists with a specific selection i splitted it to the specific selections
closed 2025-06-23 by chennes