FreeCAD/FreeCAD #21394: Core: Quantity unable to round trip parse its output

davesrocketshop, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Trying to parse the output of Quantity.UserString is unable to handle certain units, such as density. I haven't exhaustively tested what else might be failing, but density is definitely failing.

Given this test case:
```
import unittest
import FreeCAD

parseQuantity = FreeCAD.Units.parseQuantity

class QuantityRoundTripTestCases(unittest.TestCase):

    def setUp(self) -> None:
        return super().setUp()

    def tearDown(self) -> None:
        return super().tearDown()

    def testQuantityCreation(self):
        quantity = parseQuantity("2400 kg/m^3")
        self.assertEqual(quantity.UserString, "2400.00 kg/m³")

        quantity2 = parseQuantity(quantity.UserString)
        self.assertEqual(quantity2.UserString, "2400.00 kg/m³")

```
Gives the following results:
```
07:50:51  FAILED: testQuantityCreation (materialtests.TestQuantity.QuantityRoundTripTestCases.testQuantityCreation)
Traceback (most recent call last):
  File "C:\Users\dcarter\Documents\Projects\freecad-build\bin\Lib\unittest\case.py", line 58, in testPartExecutor
    yield
  File "C:\Users\dcarter\Documents\Projects\freecad-build\bin\Lib\unittest\case.py", line 634, in run
    self._callTestMethod(testMethod)
  File "C:\Users\dcarter\Documents\Projects\freecad-build\bin\Lib\unittest\case.py", line 589, in _callTestMethod
    if method() is not None:
       ^^^^^^^^
  File "C:\Users\dcarter\Documents\Projects\freecad-build\Mod\Material\materialtests\TestQuantity.py", line 19, in testQuantityCreation
    self.assertEqual(quantity2.UserString, "2400.00 kg/m³")
AssertionError: '2.40 kg/mm' != '2400.00 kg/m³'
- 2.40 kg/mm
+ 2400.00 kg/m³

07:50:51  
----------------------------------------------------------------------
Ran 16 tests in 5.49s

07:50:51  FAILED (failures=1)

```

The problem appears to be in how it's outputting the units. "m^3" becomes "m³" which parses as "mm"

### Full version info

```shell
build from source:

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41729 (Git)
Build date: 2025/05/15 13:35:38
Build type: Release
Branch: editor_tree_functions2
Hash: 39505164b2dc48f5cac73c66da3bde8eafaf664f
Python 3.12.10, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, IfcOpenShell 0.8.1.post1, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windows11
Logical DPI/Physical DPI/Pixel Ratio: 96/113.561/1.25
Installed mods: 
  * CfdOF 1.27.10
  * ExplodedAssembly
  * MaterialDB 0.0.1
  * MaterialWS 0.0.1
  * OpenTheme 2024.9.1
  * Rocket 4.1.1
  * sheetmetal 0.7.22
  * Telemetry 1.0.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hyarion, 2025-05-16:
@3x380V @bofdahof fyi

hyarion, 2025-05-16:
Yes it is problematic when both refactoring and changing behavior is done at the same time. The unicode stuff should probably only be rendered, not live in the data.

3x380V, 2025-05-16:
Change reverted in #21365.

Random rant: just looked at parser more closely... Fixable... Pity we need to keep backward compatibility. Otherwise I'd just dump all C++ implementation out and used [Pint](https://pint.readthedocs.io/en/stable/index.html) and [SymPy](https://www.sympy.org/en/index.html)

hyarion, 2025-05-17:
> Random rant: just looked at parser more closely... Fixable... Pity we need to keep backward compatibility

Yeah, it's quite annoying, but important.

It reminds me of @kadet1090 's migration proposal, that would be another way to fix it.

PR #21365 (2025-05-19): Base: fix more Unit schema management regressions
Fixes: #21317
Fixes: #21353
Fixes: #21373
Fixes: #21394
closed 2025-05-19 by hyarion