FreeCAD/FreeCAD #21396: Sketcher: MissingVerticalHorizontal functions do not behave as expected (false positive)

Roy-043, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The MissingVerticalHorizontal functions do not behave as expected (false positives).

1. Open the attached file (remove .zip).
2. Paste this in the Python console:
```
doc = App.ActiveDocument
obj = doc.Sketch

obj.detectMissingPointOnPointConstraints(0.1)
obj.makeMissingPointOnPointCoincident(False)
doc.recompute()
```
3. Result: OK. No new constraints are added.
4. Paste this in the Python console:
```
obj.detectMissingVerticalHorizontalConstraints(0.1)
obj.makeMissingVerticalHorizontal(False)
doc.recompute()
```
5. Result: error. The horizontal edge and vertical edge (that already had a horizontal or vertical constraint) now are over-constrained.

[Test-sketch-constraint.FCStd.zip](https://github.com/user-attachments/files/20243874/Test-sketch-constraint.FCStd.zip)

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41671 (Git)
Build date: 2025/05/12 16:21:14
Build type: Release
Branch: main
Hash: a997c1ef16a0a4657e0edf61fb15d01058832227
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-17:
Just to give full context here, initially the two booleans in _make_sketch.py_ were set to `False` in order to fix [#19978](https://github.com/FreeCAD/FreeCAD/issues/19978) so a Unit Test using the SVG file from that issue and the DXF file from here needs to be written in order to cover off both scenarios IMHO.

Roy-043, 2025-05-17:
Oops, I forgot about that one. Thanks for reminding.

I am not sure how tests based on existing files should be implemented. Are there examples of that in other workbenches?

I'll try to submit a new PR anyway.

Roy-043, 2025-05-17:
After reverting #21398 on my system, and testing with the scenario from #19978 I get a sketch with 4 redundant constraints. Unless my memory fails me (again 😄 ) this did not happen when testing #19986.

Roy-043, 2025-05-18:
I have found an explanation for the 8 redundant constraints: The SVG of #19978 contains 2 duplicate squares.

closed 2026-09-08 by maxwxyz