FreeCAD/FreeCAD #21400: BIM: Filtering out Level and Building objects during conversion to strict IFC mode

semhustej, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- create new file
- create `Arch_Level` object
- create `Arch_Wall` and place it in the Level
- switch the project to strict IFC mode
- the level and the wall disappear during the conversion. This is the information from the Report view
`DEBUG: Filtering out Level`

See behavior in the video:

https://github.com/user-attachments/assets/f560c9aa-84c4-4b90-b8fb-6503f69d4ca9

The same thing happens with the `Arch_Building` container.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Anything else?

This issue did not happen with the previous Windows weekly dev version with IfcOpenShell 0.8.0, it is a regression from that point:

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tetektoza, 2025-05-16:
Thanks Tom, I will start solving all of those problems this weekend, so a big wave is coming.

Can you check if it happens if you switch to ifcopenshell 0.8.0? Are you able to do that? I experience the same with ifcopenshell 0.8.1-post without any of my recent changes, so looks like a regression on their side?

semhustej, 2025-05-16:
> Thanks Tom, I will start solving all of those problems this weekend, so a big wave is coming.
>
> Can you check if it happens if you switch to ifcopenshell 0.8.0? Are you able to do that? I experience [...]

Thanks again for looking into the issue. I don't know how to test with IOS 0.8.0. Working with IOS versions is tricky.

I did not do a test with the 1.1.0dev.41365 IOS 0.8.1 version but I have been using this version for the past month, so I just guessed that I had to try to convert a project with building and level to IFC strict mode. But I am not 100% sure.

There is also an option that it might not be related to IfcOpenShell version.

closed 2025-06-23 by yorikvanhavre

semhustej, 2025-06-27:
Tested with the latest weekly and it is fixed. Thank you.
```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1

```