FreeCAD/FreeCAD #21402: BIM WebGL export: allow user to modify the HTML template

FlachyJoe, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The HTML template is currently hard-coded in [src/Mod/BIM/importers/importWebGL.py](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/importers/importWebGL.py).
If it wasn't the user could modify it or provide another one by file.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41698 (Git)
Build date: 2025/05/14 13:05:56
Build type: Unknown
Branch: main
Hash: 9fb2fbf0f7a0bf0d2a63e6bf843ede87453fa317
Python 3.13.3, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 85/86.4394/1
Installed mods: 
  * Render 2024.12.15
  * Curves.backup1744720429.3214982 0.6.60 (Disabled)
  * lattice2 1.0.0
  * Lithophane
  * Curves 0.6.62
  * fasteners 0.5.38
  * Assembly4 0.50.18
  * Telemetry 1.0.1
  * FreeCAD_SketchArch
  * Road 2025.2.22
  * FCPDWorkbench 0.4.0
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-16:
@chiragsingh1711 maybe related to your 3D PDF GSoC project?

yorikvanhavre, 2025-05-23:
That's a good idea, and probably not hard to implement