FreeCAD/FreeCAD #21405: BIM: BIM_Views panel - enable deactivation

semhustej, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**

**Problem description**
Recently, `Activate` has been added to the BIM_Views panel: https://github.com/FreeCAD/FreeCAD/pull/21239
At the moment, the user has no option to deactivate the activated object through BIM_Views panel. User has to use project tree to deactivate the object.

**Proposed feature**
To add consistency to BIM_Views panel, it would be nice if user could reverse the Activation of the object from the BIM_Views.

The proposed behavior:

- if an object is activated and user opens RMB context menu, the `Activate` option should be replaced by `Deactivate` option. This would be consistent with the behavior of project tree.

[image]

The green checkbox icon should be replaced by something else. I think either DialogClose or MediaStop icons would be OK:

[image]

[image]

- DoubleClicking on active object should deactivate the object, similar way it does in project tree

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-05-16:
@semhustej thanks. I wrote a long paragraph with my thoughts, and then I realized this pattern is already implemented in Part Design :-)

1. Double-click to activate a body
2. To deactivate it: you either double-click on another body, or uncheck the "Activated body" checkbox on the right mouse button context menu.
3. The "Active body" checkbox can also be used as an alternative to double-clicking.

[image]

My suggestion would be to implement the same UX.

Then the question is: what happens when a level is being deactivated?
- Does the working plane return to a specific default?
- If we end up implementing [this proposal](https://github.com/FreeCAD/FreeCAD/pull/21291?) (restore saved view with activation), what do we do with the view when deactivating a level?

semhustej, 2025-05-16:
> Then the question is: what happens when a level is being deactivated?
>
>     * Does the working plane return to a specific default?
>
>     * If we end up implementing [this [...]

My initial thought is I would not touch the working plane and view when deactivating level/building. I don't think there is need for that. User could have changed them even with level being active anyway.

furgo16, 2025-05-16:
> Calling Levels and Buildings Container or Body or Container

To clarify, that was just the example in Part Design. I'm not proposing to call them like that. If I were to propose something, it would be simply `Active`

furgo16, 2025-05-16:
> My initial thought is I would not touch the working plane and view when deactivating level/building. I don't think there is need for that. User could have changed them even with level being active anyway.

That's the part I don't quite understand then. If Activate simply selects the workplane currently, what should Deactivate do?

semhustej, 2025-05-16:
> That's the part I don't quite understand then. If Activate simply selects the workplane currently, what should Deactivate do?

Oh I see that current behavior is 

- `Activate` level saves the last used working plane and sets the level as an active workplane. After `Deactivate` is run, the working plane returns to working plane that was last used before activate command.
- If user activates a level and than changes the working plane (to lets say Working Plane Sied) while level is activated and and then deactivates, the Working Plane Side remains active. It does not return to the working plane before the `Activate` command.

I currently don't have a conviction on this, so I would not propose any changes. The Activate/Deactivate behavior currently works through `Project Tree`, I would leave it as it is working only applied to `BIM_Views` panel.

If a saved view is added to the `Activate` command, I don't thin I would bother saving the last used view before the `Activate` command. So after `Deactivate` nothing happens to the view.

furgo16, 2025-05-16:
That all sounds sensible to me. 

> At the moment, the user has no option to deactivate the activated object through BIM_Views panel. User has to use project tree to deactivate the object.

Actually, how do you actually do this? When rightl-clicking on a level on the tree view, I don't see either command, neither `Activate` nor `Deactivate`.

furgo16, 2025-05-16:
Oh, I see. The `DoubleClickActivates` view property needs to be set to `true` for the `Activate` and `Deactivate` command to appear on the righ mouse button context menu.

Still, I think for consistency with Part Design, it would be good if there were only an entry called `Active` with a checkbox next to it.

kaiwas, 2025-05-16:
I think we opened Pandora's box with these activations/deactivations )))
But now at least I see the light at the end of the tunnel.

furgo16, 2025-05-17:
> I think we opened Pandora's box with these activations/deactivations )))

Indeed :) I've been using the BIM workbench for over a year at least, and I'm just starting to learn about these behaviours. A couple of other thoughts:

1. In the current version of FreeCAD, how does one activate a level, if `DoubleClickActivates` is not set to `true`?
2. The more I think about it, the more I feel `Activate` should do at least two things:
   1. Set the working plane to that of the level => current behaviour
   2. "Capture" newly created objects => new, additional behaviour. Capturing meaning that when the container is activated, all new objects are created as children of it. That would make it explicit and known to the user that we're in this mode. That said, my initial expectation always was that when a container (e.g. group or level) is selected in the tree view, then all new objects are created as their children, regardless of whether active or not. Either way, active or not, the ability to set where the children are put would be a significant UX improvement.

kaiwas, 2025-05-17:
> However, this does not work for groups. Even with a level active, if you create a new group, the group will always be created at the root of the document tree.

Yes. That's why I created two issues on this topic. I think we are already very close to the ideal behavior of tree view in BIM (Provided that such changes will be made)
https://github.com/FreeCAD/FreeCAD/issues/21382
https://github.com/FreeCAD/FreeCAD/issues/21380

tetektoza, 2025-05-17:
So, what's the final decision here? We just want to switch the button once it gets activated and then pressing `Deactivate` would just put it out from active WP? We don't plan on restoring any view as discussed before, right?

kaiwas, 2025-05-18:
I think "Deactivation" is like double clicking on "active level"
maybe WP change on TOP

semhustej, 2025-06-08:
Tested with the latest weekly. Works very well, thank you.

The only thing that confused me is that next to Active is the `(default)` text in the RMB menu. At first I had a thought that the level is active by default. Although I understand the `default` probably means it is a default double-click action, it's a bit confusing.

I would suggest removing this text `(default)`. It would also lead to better consistency with the RMB menu in project tree as you can see in the screenshots. What are your thoughts?

[image]

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

furgo16, 2025-06-08:
> I would suggest removing this text (default). It would also lead to better consistency with the RMB menu in project tree as you can see in the screenshots. What are your thoughts?

I'd agree. "Default" is redundant, as the entry is in bold to indicate it's the default, and the Tree View counterpart does not show this extra text either.