FreeCAD/FreeCAD #21406: Wayland: running `FreeCAD --version --verbose` from CLI outputs wayland related warnings

luzpaz, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

```shell
$ ./FreeCAD_weekly-builds-41694-conda-Linux-x86_64-py311.AppImage --version --verbose                                                                                 
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
QThreadStorage: entry 2 destroyed before end of thread 0x5d1c8344d270
QThreadStorage: entry 1 destroyed before end of thread 0x5d1c8344d270
```

### Full version info

```shell
OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git) AppImage
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/149.623/1
Installed mods: 
  * fcVM
  * fcVM.backup1747322268.288692 (Disabled)
  * FreeCAD-Ribbon 1.9.1
  * SearchBar 1.4.2
  * Silk 0.1.8
  * Telemetry 1.0.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

adrianinsaval, 2025-05-16:
use 
```
FreeCAD_whatever.AppImage freecadcmd --version --verbose
```
or
```
FreeCAD_whatever.AppImage --cmd --version --verbose
```
for terminal only usage

luzpaz, 2025-05-17:
Thanks @adrianinsaval 
TIL

> ```
> FreeCAD_whatever.AppImage freecadcmd --version --verbose
> ```

```shell
$ ./FreeCAD_weekly-builds-41694-conda-Linux-x86_64-py311.AppImage freecadcmd --version --verbose                                                                                                                                                                                                        
OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git) AppImage
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Installed mods: 
  * fcVM
  * fcVM.backup1747322268.288692 (Disabled)
  * FreeCAD-Ribbon 1.9.1
  * SearchBar 1.4.2
  * Silk 0.1.8
  * Telemetry 1.0.1
```

Note: I get the OP issue when I run the following though:

> ```
> FreeCAD_whatever.AppImage --cmd --version --verbose
> ```

```shell
$ ./FreeCAD_weekly-builds-41694-conda-Linux-x86_64-py311.AppImage --cmd --version --verbose                                                                                                                                                                                                                      qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
QThreadStorage: entry 2 destroyed before end of thread 0x6383ddfc82a0
QThreadStorage: entry 1 destroyed before end of thread 0x6383ddfc82a0
```

luzpaz, 2025-05-21:
@adrianinsaval I also get the warning when running:

```shell
$ ./FreeCAD_weekly-builds-41726-conda-Linux-x86_64-py311.AppImage                                                                                                                                                                                                                                            
FreeCAD 1.1.0, Libs: 1.1.0devR41726 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
```