FreeCAD/FreeCAD #21410: FreeCAD_1.0.1 Appimage not loading.

VictorGallagher, 2025-05-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The latest release of FreeCAD is not loading in Linux Mint. This is the error I get when I try running it from the command line.
./FreeCAD_1.0.1-conda-Linux-x86_64-py311.AppImage
fuse: memory allocation failed
squashfuse 0.5.2 (c) 2012 Dave Vasilevsky

Usage: squashfuse [options] ARCHIVE MOUNTPOINT

(null) options:
    -o offset=N            offset N bytes into ARCHIVE to mount
    -o subdir=PATH         mount subdirectory PATH of ARCHIVE
    -o notify_pipe=PATH    named pipe that will receive 's' (success)
                           or 'f' (failure) when the mountpoint is ready
    -o timeout=N           idle N seconds for automatic unmount
    -o uid=N               set file owner to uid N
    -o gid=N               set file group to gid N

FUSE options:
    -h   --help            print help
    -V   --version         print version
    -d   -o debug          enable debug output (implies -f)
    -f                     foreground operation
    -s                     disable multi-threaded operation
    -o clone_fd            use separate fuse device fd for each thread
                           (may improve performance)
    -o max_idle_threads    the maximum number of idle worker threads
                           allowed (default: -1)
    -o max_threads         the maximum number of worker threads
                           allowed (default: 10)
fuse: memory allocation failed
squashfuse 0.5.2 (c) 2012 Dave Vasilevsky

Usage: squashfuse [options] ARCHIVE MOUNTPOINT

(null) options:
    -o offset=N            offset N bytes into ARCHIVE to mount
    -o subdir=PATH         mount subdirectory PATH of ARCHIVE
    -o notify_pipe=PATH    named pipe that will receive 's' (success)
                           or 'f' (failure) when the mountpoint is ready
    -o timeout=N           idle N seconds for automatic unmount
    -o uid=N               set file owner to uid N
    -o gid=N               set file group to gid N

FUSE options:
    -h   --help            print help
    -V   --version         print version
    -d   -o debug          enable debug output (implies -f)
    -f                     foreground operation
    -s                     disable multi-threaded operation
    -o clone_fd            use separate fuse device fd for each thread
                           (may improve performance)
    -o max_idle_threads    the maximum number of idle worker threads
                           allowed (default: -1)
    -o max_threads         the maximum number of worker threads
                           allowed (default: 10)
squashfuse 0.5.2 (c) 2012 Dave Vasilevsky

Usage: squashfuse [options] ARCHIVE MOUNTPOINT

(null) options:
    -o offset=N            offset N bytes into ARCHIVE to mount
    -o subdir=PATH         mount subdirectory PATH of ARCHIVE
    -o notify_pipe=PATH    named pipe that will receive 's' (success)
                           or 'f' (failure) when the mountpoint is ready
    -o timeout=N           idle N seconds for automatic unmount
[... 109 lines cut ...]
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/teejee2008-foss-jammy.list
    1: deb http: //ppa.launchpad.net/teejee2008/foss/ubuntu jammy main
  Active apt repos in: /etc/apt/sources.list.d/vivaldi.list
    1: deb [arch=amd64] https: //repo.vivaldi.com/stable/deb/ stable main
Info:
  Processes: 371 Uptime: 6h 31m Memory: 46.18 GiB used: 3.83 GiB (8.3%) Init: systemd v: 249
  runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12 Client: Cinnamon v: 5.4.12 inxi: 3.3.13

The command works fine for the previous stable version.

### Full version info

```shell
FreeCAD_1.0.1-conda-Linux-x86_64-py311.AppImage
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-05-17:
Can no reproduce in Archlinux

https://github.com/user-attachments/assets/b0d734a4-6154-4d87-beb1-42bae9c4a990

Syres916, 2025-05-17:
It works using Linux Mint 21.3 which I believe shipped with the more up-to-date AppImageLauncher. I notice @VictorGallagher has the AppImageLauncher PPA in the report:

```
Active apt repos in: /etc/apt/sources.list.d/appimagelauncher-team-stable-jammy.list
1: deb http: //ppa.launchpad.net/appimagelauncher-team/stable/ubuntu jammy main
```

According to the AppImageLauncher Wiki the PPAs have become [unmaintained](https://github.com/TheAssassin/AppImageLauncher/wiki/Install-on-Ubuntu-or-Debian#use-the-ppas) and broken so the first thing I would do is remove that from `Software Sources`

The current instructions are to download https://github.com/TheAssassin/AppImageLauncher/releases/download/v2.2.0/appimagelauncher_2.2.0-travis995.0f91801.bionic_arm64.deb to `/tmp` folder

Then run:

```
wget https://github.com/TheAssassin/AppImageLauncher/releases/download/v2.2.0/appimagelauncher_2.2.0-travis995.0f91801.bionic_amd64.deb -O /tmp/appimagelauncher.deb
sudo dpkg -i /tmp/appimagelauncher.deb
```

So you know that you're running the latest stable version and report back if it still does not launch.

VictorGallagher, 2025-05-17:
Solution worked. Thanks!

closed 2025-05-17 by VictorGallagher

papylhomme, 2025-06-03:
Hello, same problem using Manjaro, but appimagelauncher is already at version 2.2.0
> AppImageLauncher version 2.2.0 (git commit 9ca001e), built on 2025-06-02 22:07:46 UTC

@tarman3 what was your version on arch ?