FreeCAD/FreeCAD #21419: Draft_Shape2DView shows QuickMeasure error

semhustej, 2025-05-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file
- create `Draft_Line`
- select the line and invoke Draft_Shape2DView
- error shows up with no visible problems
`<QuickMeasure> QuickMeasure.cpp(98): 0000020ECC678340 : Standard_TypeMismatch: TopoDS::Edge`

See behavior in the video:

https://github.com/user-attachments/assets/be94bfea-b66d-4989-a611-ae2c067b8e0e

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
```

### Subproject(s) affected?

Draft

### Anything else?

Might be related to https://github.com/FreeCAD/FreeCAD/issues/19579

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-17:
At first I thought this might have been because the Shape2DView object was directly on top of the line but it's caused by the type of selection. If the user selects the Shape2DView object in the 3D View the selection is shown in the Python Console as:

`Gui.Selection.addSelection('Unnamed','Shape2DView','Edge1',-245.605,-78.9721,0)`

and no error is generated but if the user does the same but in the Tree view the selection is shown in the Python Console as:

`Gui.Selection.addSelection('Unnamed','Shape2DView')`

and the `<QuickMeasure> QuickMeasure.cpp(98): 0x5d6e56f58000 : Standard_TypeMismatch: TopoDS::Edge` error is generated.

Both `Line` and `Shape2DView` objects show `isDerivedFrom` as `Part::Part2DObjectPython`

Syres916, 2025-05-19:
Running a bisect from end of October through to early December 2024 as that was the best I could narrow it down, will report back later today.

Syres916, 2025-05-19:
Bisect determines https://github.com/FreeCAD/FreeCAD/commit/df3bf98ef784f8a4e3bb86eda93753d54cd39ef8 is responsible.

Syres916, 2025-05-19:
In version 1.0.x a _Draft Line_, _Shape2DView_ and a _Sketch with a single line_ are all incorrectly detected as `MeasureType::Volumes` so the commit above was meant to detect _Solids_ correctly as `MeasureType::Volumes` fixing `Part::Lines` but I need to determine why _Shape2DView_ objects are dropping through the various criteria to `TypeMismatch`.

ErtugrulEge, 2025-05-19:
also same message with arch axis 

https://github.com/freecad/freecad/issues/19579

Syres916, 2025-05-19:
@PaddleStroke below are the types for Draft Line, Shape2dView of previously created Draft Line, Sketch with a single line and a Arch Axis. Could you please create a fix to handle these correctly:

```
Draft Line
14:50:34  case TopAbs_WIRE
14:50:34  It is MeasureType::Edges
14:50:34  <QuickMeasure> QuickMeasure.cpp(97): 0x5c5ac190d400 : Standard_TypeMismatch: TopoDS::Edge
```

```
Shape2dView of above Draft Line
14:50:46  case TopAbs_EDGE
14:50:46  It is MeasureType::Line
14:50:46  <QuickMeasure> QuickMeasure.cpp(97): 0x5c5ac190d400 : Standard_TypeMismatch: TopoDS::Edge
```

```
Sketch
14:51:58  case TopAbs_WIRE
14:51:58  It is MeasureType::Edges
14:51:58  <QuickMeasure> QuickMeasure.cpp(97): 0x5c5ac190d400 : Standard_TypeMismatch: TopoDS::Edge
```

```
Arch_Axis
14:55:12  case TopAbs_EDGE
14:55:12  It is MeasureType::Line
14:55:12  <QuickMeasure> QuickMeasure.cpp(97): 0x555c710a7f80 : Standard_TypeMismatch: TopoDS::Edge
```

Syres916, 2025-05-20:
@hlorus FYI

Roy-043, 2025-07-03:
AFAICT this has been fixed in the current dev version, and the fix has also been backported to V1.0.1.

ErtugrulEge, 2025-07-04:
I have no errors with latest weekly 

```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42461 (Git)
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

Roy-043, 2025-07-04:
Thanks. Closing then.

closed 2025-07-04 by Roy-043

Syres916, 2025-07-24:
@Roy-043 for completeness it was https://github.com/FreeCAD/FreeCAD/pull/21257/commits/9c5c6c9af2c59edd0cc1145ec748f064814c3887 that inadvertently fixed this issue, flushed out by https://github.com/FreeCAD/FreeCAD/issues/22686 and the PR to fix it https://github.com/FreeCAD/FreeCAD/pull/22691