FreeCAD/FreeCAD #21420: addon manager blocked from wiki

aavogt, 2025-05-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

in the addon manager (`M-t  M-a`) click on an extension that has (wiki) after the name. If I click "3d printer slicer (wiki)" I should get either:

[image]

[image]

Instead it gets stuck at with a misleading message:

[image]

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (default/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/05/13 09:39:01
Build type: Release
Branch: main
Hash: 7d57569eb493b929e5647bd9b0ad03b33d1acc45
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
Installed mods: 
  * CfdOF 1.29.4
  * sheetmetal 0.6.13
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

aavogt, 2025-05-17:
Even after I close the addon manager, freecad seems to make requests roughly 6 times per second:
```
11:36:07  (qt.network.http2) stream 2001 finished with error: "Server stopped accepting new streams before this stream was established"
11:36:07  (qt.network.http2) stream 2003 finished with error: "Server stopped accepting new streams before this stream was established"
11:36:07  Request failed: NetworkError.ContentReSendError 
11:36:07  Addon Manager: Unexpected 0 response from server
11:36:07  Unable to open macro wiki page at https://wiki.freecad.org/Macro_3d_Printer_Slicer
11:36:07  Request failed: NetworkError.ContentReSendError 
11:36:07  Addon Manager: Unexpected 0 response from server
11:36:07  Unable to open macro wiki page at https://wiki.freecad.org/Macro_3d_Printer_Slicer_Individual_Parts
11:36:07  Unable to fetch the code of this macro.
```

```
> grep Unable /tmp/log.txt
11:53:20  Unable to fetch the code of this macro.
11:53:20  Unable to fetch the code of this macro.
11:53:20  Unable to fetch the code of this macro.
11:53:20  Unable to fetch the code of this macro.
11:53:20  Unable to fetch the code of this macro.
11:53:20  Unable to fetch the code of this macro.
11:53:20  Unable to fetch the code of this macro.
11:53:21  Unable to fetch the code of this macro.
11:53:21  Unable to fetch the code of this macro.
11:53:21  Unable to fetch the code of this macro.
11:53:21  Unable to fetch the code of this macro.
11:53:21  Unable to fetch the code of this macro.
11:53:21  Unable to fetch the code of this macro.
11:53:21  Unable to fetch the code of this macro.
11:53:21  Unable to fetch the code of this macro.
11:53:21  Unable to fetch the code of this macro.
11:53:22  Unable to fetch the code of this macro.
11:53:22  Unable to fetch the code of this macro.
11:53:22  Unable to fetch the code of this macro.
11:53:22  Unable to fetch the code of this macro.
11:53:22  Unable to fetch the code of this macro.
11:53:22  Unable to fetch the code of this macro.
11:53:22  Unable to fetch the code of this macro.
11:53:22  Unable to fetch the code of this macro.
11:53:23  Unable to fetch the code of this macro.
11:53:23  Unable to fetch the code of this macro.
11:53:23  Unable to fetch the code of this macro.
11:53:23  Unable to fetch the code of this macro.
11:53:23  Unable to fetch the code of this macro.
11:53:23  Unable to fetch the code of this macro.
11:53:24  Unable to fetch the code of this macro.
11:53:24  Unable to fetch the code of this macro.
11:53:24  Unable to fetch the code of this macro.
11:53:24  Unable to fetch the code of this macro.
11:53:24  Unable to fetch the code of this macro.
11:53:24  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:25  Unable to fetch the code of this macro.
11:53:26  Unable to fetch the code of this macro.
11:53:26  Unable to fetch the code of this macro.
11:53:26  Unable to fetch the code of this macro.
```

Roy-043, 2025-05-17:
Duplicate of #21122.

closed 2025-05-17 by Roy-043