FreeCAD/FreeCAD #21421: BIM: Better way to pass the Classification during BIM Object (Standard Code) -> IFC Object conversion

semhustej, 2025-05-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Created as per comment by @tetektoza :https://github.com/FreeCAD/FreeCAD/pull/21243#issuecomment-2888422887

Better way to pass the Classification during BIM Object (Standard Code) -> IFC Object conversion, or IFC Object -> IFC Type conversion:

_Basically, during add_properties I've added additional handling to actually get the ifcrelassociateclassification list, which holds references to the object that has classification + it's classification reference. This way it's much easier to see if the currently processed entity actually needs the property being assigned. If it misses the property, and is on the ifcrelassociateclassification list, it means that it's a newly created object from conversion (IFC obj or IFC Type), and it previously either had Standard Code or Classification assigned, so we add the property to the newly created object. If the entity is not on the list in any of referenced classifications, this means that this is just a standard, newly created IFC Object, so we don't assign anything to it. Much cleaner than the if statement in assign_classification and the hack with having the newly generated object in the proxy object._

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tetektoza, 2025-05-17:
I think we misunderstood each other, I replied regarding this:
https://github.com/FreeCAD/FreeCAD/pull/21243#issuecomment-2881047386

I've said that this is a good idea, to expand Classification to separate group and add more parameters :D

semhustej, 2025-05-18:
> I think we misunderstood each other, I replied regarding this: [#21243 (comment)](https://github.com/FreeCAD/FreeCAD/pull/21243#issuecomment-2881047386)
> 
> I've said that this is a good idea, to expand Classification to separate group and add more parameters :D

Ah, I was a bit puzzled about the request :). Closing this one.

closed 2025-05-18 by semhustej