FreeCAD/FreeCAD #21422: BIM: FC 1.0.1 'typing_extensions' error during multiple basic actions

semhustej, 2025-05-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using BIM workbench, current stable FreeCAD 1.0.1 posts 'typing_extensions' error during multiple basic actions. The functionality is not influenced.

**Steps to reproduce option 01**

- open a new project
- create `Arch_Wall` (or any other BIM object - tested with Slab, Window)
- delete the wall
- error shows up

```
18:40:05  pyException: Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\Mod\BIM\nativeifc\ifc_observer.py", line 71, in slotDeletedObject
    from nativeifc import ifc_tools  # lazy loading
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\Mod\BIM\nativeifc\ifc_tools.py", line 36, in <module>
    from ifcopenshell import geom
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\ifcopenshell\geom\occ_utils.py", line 27, in <module>
    from typing_extensions import assert_never
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'typing_extensions'

```

**Steps to reproduce option 02**

- open new file
- activate BIM workbench
- create Draft_Line (or Draft_Wire)
- error shows up

```
18:41:53  pyException: Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\Mod\BIM\nativeifc\ifc_observer.py", line 71, in slotDeletedObject
    from nativeifc import ifc_tools  # lazy loading
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\Mod\BIM\nativeifc\ifc_tools.py", line 36, in <module>
    from ifcopenshell import geom
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\ifcopenshell\geom\occ_utils.py", line 27, in <module>
    from typing_extensions import assert_never
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'typing_extensions'

```

When creating Draft_Line in Draft WB, no error shows up. Only when BIM WB is activated.

See behavior in a video:

https://github.com/user-attachments/assets/06125f78-9a2a-4c1c-811f-1bc97421b623

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Anything else?

One of the issues has been discovered by a user in this thread: https://forum.freecad.org/viewtopic.php?t=97121

It is probably related to this issue: https://github.com/FreeCAD/FreeCAD/issues/19051

If the issue is confirmed, this would mean that basically every BIM WB user will get lot of these errors on Windows using the current stable release.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-05-17:
I did a test in Safe Mode with the same results.

ErtugrulEge, 2025-05-17:
I can confirm
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38
```

Syres916, 2025-05-17:
At least it seems consistent across platforms so shouldn't be difficult to add it to the packages across the board for the next release FYI @adrianinsaval . As a workaround I ran the following as a macro:

```
module_to_install = "typing-extensions"
import os
import FreeCAD
import addonmanager_utilities as utils
vendor_path = utils.get_pip_target_directory()
cmd = []
args = ["install", "--disable-pip-version-check", "--target", vendor_path, module_to_install, "--upgrade"]
snap_package = os.getenv("SNAP_REVISION")
appimage = os.getenv("APPIMAGE")
if snap_package:
    index = args.index("--target")
    del args[index : index + 2]  # The --target option and its argument
    cmd = ["pip"]
elif appimage:
    python_exe = App.getHomePath() + "bin/python"
else:
    if hasattr(utils, "get_python_exe"):
        python_exe = utils.get_python_exe()
    else:
        from freecad.utils import get_python_exe
        python_exe = get_python_exe()
# Just incase someone has extracted an AppImage
if "squashfs-root" in App.getHomePath():
    python_exe = App.getHomePath() + "bin/python"
if not cmd:
    cmd = [python_exe, "-m", "pip"]
cmd.extend(args)
result = None
try:
    result = utils.run_interruptable_subprocess(cmd)
    FreeCAD.Console.PrintMessage("Install/Upgrade of " + module_to_install + " finished\n")
except:
    text = "Error trying to Install/Upgrade " + module_to_install + ". Please ensure pip is installed on your system and the module exists for this version of Python"
    FreeCAD.Console.PrintError(text + "\n")
```

Notice how the module is called `typing-extensions` but the module to import is `typing_extensions` (underscore instead of hyphen)

```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark_Pre1_0.qss/unset/Qt default
Installed mods: 
  * CfdOF 1.29.8
  * sheetmetal 0.7.10
```

adrianinsaval, 2025-05-17:
Aaagg I forgot about this one, I will rebuild the release including this module tomorrow or monday

markit, 2025-05-18:
> Aaagg I forgot about this one, I will rebuild the release including this module tomorrow or monday

Could you please also fix it for 1.1dev (weekly releases)?
See i.e. bug https://github.com/FreeCAD/FreeCAD/issues/19051

ErtugrulEge, 2025-06-03:
Now, I downloaded stable version and weekly versions. They are working fine.Please, try and confirm.

https://github.com/FreeCAD/FreeCAD/issues/19051

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Installed mods: 
  * Curves 0.6.61
```

semhustej, 2025-06-19:
Is there a plan to release FC 1.0.2 in the near future with this issue fixed?

I see the issue is added to the 1.1 milestone, which in my opinion does not seem to capture the importance of this issue. I think this is quite a serious regression.

The issue means that every user of the BIM workbench is exposed to error messages every time a BIM command is invoked, at least that's what I understand. This renders the BIM workbench unusable without some tweaking or manually installing typing_extensions.

With this issue I don't think we can expect any new BIM user to be exposed to FreeCAD without thinking negative things about the software and probably resigning on the software. BIM is one of the core workbenches.

ErtugrulEge, 2025-06-19:
It is very bad for the new user to encounter an error when they first draw a simple line or wall.I can't say anything for the other issues , but it would be great if this bug was fixed sooner, like in 1.0.2

semhustej, 2025-07-14:
@tbuegel I am not a developer, so I can not really help with that.  I don't know if another python version installed might cause the problem.

I would try to backup the existing `typing_extensions.py` and remove it from to location and then run the python command again.

There is also this post from Roy_043, which seems contain more refined script which you might run: https://forum.freecad.org/viewtopic.php?p=825237#p825237

furgo16, 2025-07-19:
https://github.com/FreeCAD/FreeCAD-developer-meetings/blob/main/Minutes/minutes-2025-07-19.md

semhustej, 2025-07-19:
Thanks @furgo16 !

PeterBramwell, 2025-07-23:
Thanks Semhustej
Your python script worked for me in Windows
I was about to give up on FreeCAD --- after Sketchup pro 2017 it is very different!!!!
Thanks again
Peter 
PS couldn't find any way to parameterize sketchup files - such is life - will have to start from scratch again - boo hoo!

furgo16, 2025-07-31:
@adrianinsaval @chennes @yorikvanhavre I just learnt that there are plans to do a point release tomorrow (which might be today already): https://github.com/FreeCAD/FreeCAD/pull/22849#issuecomment-3138238511.

Given this was a packaging fix, I understand that the new point release will automatically include it when the package is rebuilt, correct? Or is there anything else to backport? Thank you.

furgo16, 2025-07-31:
Thanks. The snap in particular has never had this issue, as it pulls Ifcopenshell from pip, which installs all dependencies automatically.

closed 2025-08-07 by semhustej