FreeCAD/FreeCAD #21423: Sketcher: add ability to trim external projection geometry

eblanshey, 2025-05-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

A common pattern is to create a high-level master sketch, and create other sketches based on it, with slight modifications. When creating external geometries that reference other sketches, it is not possible to trim those lines. This is especially problematic for complex lines, namely B-splines, which are difficult to "trace over". It would be great to be able to support this.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/05/13 09:39:01
Build type: Release
Branch: main
Hash: 7d57569eb493b929e5647bd9b0ad03b33d1acc45
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * WebTools 1.0.0
  * Curves 0.6.58
  * Plot 2024.11.26
  * AirPlaneDesign 0.4.1
  * Help 1.0.3
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-05-17:
This should be for intersection geometry as well. FYI @AjinkyaDahale has a FPA grant to do exactly this (and more).

luzpaz, 2025-05-17:
@AjinkyaDahale keep open ?

AjinkyaDahale, 2025-05-17:
Yes.

kaiwas, 2025-05-17:
Need some kind of converter tool that will convert external geometry into normal one.