FreeCAD/FreeCAD #21429: Part: revolving Axis or Datum Line triggers crash

ElementW, 2025-05-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Applying a `Part_Revolve` on a coordinate system's axis or a Datum Line triggers an instant crash.
As little as doing this makes sense (happened to me due to UI misinterpretation), it should not crash, instead failing like it does with datum points and datum planes.

### Full version info

```shell
OS: Void Linux (KDE/wayland)
Architecture: x86_64
Version: 1.1.0dev.41726 (Git)
Build date: 2025/05/16 17:15:43
Build type: Release
Branch: main
Hash: 1ca108f8d503b2b68455844bfdfabb62fa1461ed
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.38
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.9.0.1
  * SearchBar 1.0.2
```

### Subproject(s) affected?

Part

### Anything else?

<details>
<summary>Stack trace</summary>

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x40110) [0x7f54a5069110]
#1  0x7f54780c789b in BRepSweep_Rotation::IsInvariant(TopoDS_Shape const&) const from /tmp/.mount_FreeCAHEegIH/usr/lib/././libTKPrim.so.7.8+0x2b
#2  0x7f54780c79a8 in BRepSweep_Rotation::IsInvariant(TopoDS_Shape const&) const from /tmp/.mount_FreeCAHEegIH/usr/lib/././libTKPrim.so.7.8+0x138
#3  0x7f54780d10db in BRepSweep_Trsf::Process(TopoDS_Shape const&, Sweep_NumShape const&) from /tmp/.mount_FreeCAHEegIH/usr/lib/././libTKPrim.so.7.8+0x6b
#4  0x7f54780d15cf in BRepSweep_Trsf::Init() from /tmp/.mount_FreeCAHEegIH/usr/lib/././libTKPrim.so.7.8+0x7f
#5  0x7f54780c66da in BRepSweep_Rotation::BRepSweep_Rotation(TopoDS_Shape const&, Sweep_NumShape const&, TopLoc_Location const&, gp_Ax1 const&, double, bool) from /tmp/.mount_FreeCAHEegIH/usr/lib/././libTKPrim.so.7.8+0x9a
#6  0x7f54780c5bcc in BRepSweep_Revol::BRepSweep_Revol(TopoDS_Shape const&, gp_Ax1 const&, double, bool) from /tmp/.mount_FreeCAHEegIH/usr/lib/././libTKPrim.so.7.8+0xdc
#7  0x7f54780e252b in BRepPrimAPI_MakeRevol::BRepPrimAPI_MakeRevol(TopoDS_Shape const&, gp_Ax1 const&, double, bool) from /tmp/.mount_FreeCAHEegIH/usr/lib/././libTKPrim.so.7.8+0x6b
#8  0x7f544e731815 in Part::TopoShape::makeElementRevolve(Part::TopoShape const&, gp_Ax1 const&, double, char const*, char const*) from /tmp/.mount_FreeCAHEegIH/usr/lib/./Part.so+0xa5
#9  0x7f544e4b0698 in Part::Revolution::execute() from /tmp/.mount_FreeCAHEegIH/usr/lib/./Part.so+0x2c8
#10  0x7f54a74aee92 in App::DocumentObject::recompute() from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libFreeCADApp.so+0xc2
#11  0x7f544e4c5c8c in Part::Feature::recompute() from /tmp/.mount_FreeCAHEegIH/usr/lib/./Part.so+0xc
#12  0x7f54a748b5dc in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libFreeCADApp.so+0x32c
#13  0x7f54a7492534 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libFreeCADApp.so+0x624
#14  0x7f540ed1e2cb in PartGui::DlgRevolution::accept() from /tmp/.mount_FreeCAHEegIH/usr/lib/./PartGui.so+0xcab
#15  0x7f540ed18f01 in PartGui::TaskRevolution::accept() from /tmp/.mount_FreeCAHEegIH/usr/lib/./PartGui.so+0x11
#16  0x7f54a8b25d2c in Gui::TaskView::TaskView::accept() from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libFreeCADGui.so+0x6c
#17  /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7f54a571684a]
#18  /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6(+0x3d86f9) [0x7f54a69d86f9]
#19  /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7f54a571684a]
#20  0x7f54a6a56727 in QAbstractButton::clicked(bool) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6+0x47
#21  /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6(+0x47be80) [0x7f54a6a7be80]
#22  /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6(+0x4adba7) [0x7f54a6aadba7]
#23  0x7f54a6aadd6d in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6+0xbd
#24  0x7f54a6bc3864 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6+0xa64
#25  0x7f54a6bb3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6+0x88
#26  0x7f54a6bb4cbd in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6+0xdad
#27  0x7f54a87598f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libFreeCADGui.so+0xc9
#28  0x7f54a575fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Core.so.6+0x17a
#29  0x7f54a6bbbe7c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6+0x1ec
#30  /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6(+0x5be8c7) [0x7f54a6bbe8c7]
#31  /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6(+0x5c5440) [0x7f54a6bc5440]
#32  0x7f54a6bb3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Widgets.so.6+0x88
#33  0x7f54a87598f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libFreeCADGui.so+0xc9
#34  0x7f54a575fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Core.so.6+0x17a
#35  0x7f54a5dd1970 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Gui.so.6+0x810
#36  0x7f54a5e31e14 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Gui.so.6+0xb4
#37  /tmp/.mount_FreeCAHEegIH/usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x58f90) [0x7f54a042bf90]
#38  /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/./libglib-2.0.so.0(+0x5b5e6) [0x7f54a3dbf5e6]
#39  /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/./libglib-2.0.so.0(+0x5e7b7) [0x7f54a3dc27b7]
#40  /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f54a3dc2ece]
#41  0x7f54a54f6b12 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Core.so.6+0x72
#42  0x7f54a579747c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Core.so.6+0x25c
#43  0x7f54a57977da in QCoreApplication::exec() from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libQt6Core.so.6+0x9a
#44  0x7f54a86b4676 in Gui::Application::runApplication() from /tmp/.mount_FreeCAHEegIH/usr/bin/../lib/libFreeCADGui.so+0x756
#45  /tmp/.mount_FreeCAHEegIH/usr/bin/freecad(+0x6d34) [0x55dc34a27d34]
#46  /usr/lib/libc.so.6(+0x2abfc) [0x7f54a5053bfc]
#47  /usr/lib/libc.so.6(__libc_start_main+0x85) [0x7f54a5053cb5]
#48  /tmp/.mount_FreeCAHEegIH/usr/bin/freecad(+0x703b) [0x55dc34a2803b]
```

</details>

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-17:
@ElementW can you provide an example file to trigger the crash ?

ElementW, 2025-05-17:
Sure, but honestly it's so incredibly trivial and a consistent crash the bug description is enough by itself.

[woops.FCStd.zip](https://github.com/user-attachments/files/20268431/woops.FCStd.zip)

luzpaz, 2025-05-17:
Confirmed

<details>
<summary>Stack trace</summary>

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3dcd0) [0x7ede59a5ecd0]
#1  0x7ede2408d89b in BRepSweep_Rotation::IsInvariant(TopoDS_Shape const&) const from /tmp/.mount_FreeCAGoNCEK/usr/lib/././libTKPrim.so.7.8+0x2b
#2  0x7ede2408d9a8 in BRepSweep_Rotation::IsInvariant(TopoDS_Shape const&) const from /tmp/.mount_FreeCAGoNCEK/usr/lib/././libTKPrim.so.7.8+0x138
#3  0x7ede240970db in BRepSweep_Trsf::Process(TopoDS_Shape const&, Sweep_NumShape const&) from /tmp/.mount_FreeCAGoNCEK/usr/lib/././libTKPrim.so.7.8+0x6b
#4  0x7ede240975cf in BRepSweep_Trsf::Init() from /tmp/.mount_FreeCAGoNCEK/usr/lib/././libTKPrim.so.7.8+0x7f
#5  0x7ede2408c6da in BRepSweep_Rotation::BRepSweep_Rotation(TopoDS_Shape const&, Sweep_NumShape const&, TopLoc_Location const&, gp_Ax1 const&, double, bool) from /tmp/.mount_FreeCAGoNCEK/usr/lib/././libTKPrim.so.7.8+0x9a
#6  0x7ede2408bbcc in BRepSweep_Revol::BRepSweep_Revol(TopoDS_Shape const&, gp_Ax1 const&, double, bool) from /tmp/.mount_FreeCAGoNCEK/usr/lib/././libTKPrim.so.7.8+0xdc
#7  0x7ede240a852b in BRepPrimAPI_MakeRevol::BRepPrimAPI_MakeRevol(TopoDS_Shape const&, gp_Ax1 const&, double, bool) from /tmp/.mount_FreeCAGoNCEK/usr/lib/././libTKPrim.so.7.8+0x6b
#8  0x7ede04931815 in Part::TopoShape::makeElementRevolve(Part::TopoShape const&, gp_Ax1 const&, double, char const*, char const*) from /tmp/.mount_FreeCAGoNCEK/usr/lib/./Part.so+0xa5
#9  0x7ede046b0698 in Part::Revolution::execute() from /tmp/.mount_FreeCAGoNCEK/usr/lib/./Part.so+0x2c8
#10  0x7ede5beaf002 in App::DocumentObject::recompute() from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libFreeCADApp.so+0xc2
#11  0x7ede046c5c8c in Part::Feature::recompute() from /tmp/.mount_FreeCAGoNCEK/usr/lib/./Part.so+0xc
#12  0x7ede5be8b1ac in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libFreeCADApp.so+0x32c
#13  0x7ede5be926b4 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libFreeCADApp.so+0x624
#14  0x7ede5d288dd8 in Gui::TaskTransformDialog::accept() from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libFreeCADGui.so+0x88
#15  0x7ede5d5267ac in Gui::TaskView::TaskView::accept() from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libFreeCADGui.so+0x6c
#16  /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7ede5a11684a]
#17  /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6(+0x3d86f9) [0x7ede5b3d86f9]
#18  /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7ede5a11684a]
#19  0x7ede5b456727 in QAbstractButton::clicked(bool) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6+0x47
#20  /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6(+0x47be80) [0x7ede5b47be80]
#21  /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6(+0x4adba7) [0x7ede5b4adba7]
#22  0x7ede5b4add6d in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6+0xbd
#23  0x7ede5b5c3864 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6+0xa64
#24  0x7ede5b5b3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6+0x88
#25  0x7ede5b5b4cbd in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6+0xdad
#26  0x7ede5d159e89 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libFreeCADGui.so+0xc9
#27  0x7ede5a15fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Core.so.6+0x17a
#28  0x7ede5b5bbe7c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6+0x1ec
#29  /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6(+0x5be8c7) [0x7ede5b5be8c7]
#30  /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6(+0x5c5440) [0x7ede5b5c5440]
#31  0x7ede5b5b3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Widgets.so.6+0x88
#32  0x7ede5d159e89 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libFreeCADGui.so+0xc9
#33  0x7ede5a15fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Core.so.6+0x17a
#34  0x7ede5a7d1970 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Gui.so.6+0x810
#35  0x7ede5a831e14 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/libQt6Gui.so.6+0xb4
#36  /tmp/.mount_FreeCAGoNCEK/usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x58f90) [0x7ede54c6cf90]
#37  /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/./libglib-2.0.so.0(+0x5b5e6) [0x7ede587bf5e6]
#38  /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/./libglib-2.0.so.0(+0x5e7b7) [0x7ede587c27b7]
#39  /tmp/.mount_FreeCAGoNCEK/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7ede587c2ece]
[... cut ...]
Architecture: x86_64
Version: 1.1.0dev.41694 (Git) AppImage
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Installed mods: 
  * fcVM
  * fcVM.backup1747322268.288692 (Disabled)
  * FreeCAD-Ribbon 1.9.1
  * SearchBar 1.4.2
  * Silk 0.1.8
  * Telemetry 1.0.1
```

ElementW, 2026-02-07:
@luzpaz Seems I can't trigger this from the UI any more (on 1.2.0dev.20260204), but importing the file in https://github.com/FreeCAD/FreeCAD/issues/21429#issuecomment-2888632755 still results in a crash upon recompute, meaning the underlying problem hasn't been fixed.

closed 2026-02-27 by kadet1090