FreeCAD/FreeCAD #21430: CAM: Chamfer vs V-Bit shape inconsistencies

knipknap, 2025-05-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Right now, the CAM library comes with two virtually identical shapes:

- V-Bit
- Chamfer

The only difference between the two:

- The V-Bit allows for configuring a "crest", i.e. a "crown" between the cutting edge. If this were set to 0, then it is identical to Chamfer
- In the Chamfer shape, the angle specifies the "tip angle / 2", whereas the V-Bit specifies "tip angle"

There are additional inconsistencies:

- Chamfer has a Diameter attribute, but it is unused. Only ShankDiameter is used.
- V-Bit has a CuttingEdgeHeight, but it is used in a misleading manner: It actually specifies the crown height, not the cutting edge height.

## Visual comparison

V-Bit:
[image]

Chamfer:
[image]

### Full version info

```shell
1.1dev
```

### Subproject(s) affected?

CAM

### Anything else?

## Solutions?

IMO we should:

1. Drop the chamfer bit and use V-Bit instead, because the V-Bit shape is more generic and can represent any chamfer shape already.
2. Rename CuttingEdgeHeight to CrownHeight. After #21425 this can easily be done in a backward-compatible manner.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

knipknap, 2025-05-27:
@jffmichi Makes sense, that is a more consistent fix.

bombela, 2025-09-18:
The naming is not consistent anywhere. Across manufacturers, users, and even countries (all the while conversing in English).

In a idealized world, you would model the shape of the tool in 3D, and the software would magically know how to use it by having a full understanding of the physical world. The other end of the spectrum, is one hard-coded tool shape per combination. A reasonable solution probably lies in between.

For what it worth, with 9 drivings parameters it seems possible to represent quite a large range of angled cutters.

[image]

If you wonder what is going on, I am using a macro to apply the rows of the bottom spreadsheet to the top spreadsheet, copying, and placing the shape.

The values that would make the sketcher complain are fudged by 1 nanometer (or 1 nano°). Is that one possible reason there is currently a few different V-bite shapes? Because its quite difficult of not impossible to conditionally delete geometry?

In fact I also had to order the rows of the bottom spreadsheet as to avoid the sketcher getting stuck into an unsolvable situation. Because the order in which the columns are applied can break the solver (because some geometry becomes impossible). This is a problem I very frequently encounter when generating families of parts.

Another annoyance in my example is the crown height, which is easy to set when the crown is 0° (vertical), but requires to do some math to calculate. I am sure a different driving parameter could help here. Maybe a top of crown diameter instead of the crown height. As a symmetry to the bottom angle.

We are trying to tease apart the constraints to find the right level of abstraction. Let me know if you want me to explore further.

jffmichi, 2025-09-18:
Do we really want three different bits that all look the same? Just to be able to specify them using different parameters. Also from the name you will probably not be able to tell which parameters are used inside so you will have to try them all to eventually find the one you're looking for.

How about we provide only one chamfer bit and let the user explicitly specify which parameters they want to use. Similar to how we already do it e.g. with PartDesign LinearPattern. There the user has a combobox and can choose if they want to specify the pattern by total length or by spacing. I think this should already be possible using expressions.

[image]

bombela, 2025-09-18:
> I don't think this will work. The LinearPattern is a programmed feature. The part shapes are just FreeCAD documents with parametric sketches. There's no logic in the document to switch parameter control systems. Maybe this is possible in the future with Pieter's variant part stuff but not possible now.
> 
> Also, wouldn't help much with images that look like what the user is trying to find.

You can model multiple bodies, with sketches driven conditionally based on some mode selection. Then using a subshapebinder or a link, conditionally select the shape to present.

In response to your previous post.

The number of different shape is an issue for the devs. Not the user. The user doesn't really need to care that 5 different shapes are implemented using one heavily parametrized shape or vice-versa. Click on the right picture as you demonstrated.

If having a single shape with 9 parameters means the code is easier to write and maintain, and so the operations actually do what they are supposed to do. This is technically a win in my book. Because right now, the issue is the code handles the v-bit vs chamfer-bit haphazardly.

So we have the shape file. The conceptual shape cutter. And the toolpath generation.

Maybe we have multiple shape files backing a single conceptual V-bit cutter? In fact, the code right now tries to pretend this is the case. And I think this is why the height value is miss-used in some cases as @MTronics pointed out.

jffmichi, 2025-09-18:
> I don't think this will work.

[chamfer_proof_of_concept.zip](https://github.com/user-attachments/files/22415810/chamfer_proof_of_concept.zip)

It can be done using PropertyEnumeration. A little fiddly to get the formulas right but it's possible.

sliptonic, 2025-09-19:
Nice!  The proof of concept is cool.  It's going in the direction I was referring to with variant parts. 

If I understand this correctly,  it changes which parameters are used depending on the mode. However, this doesn't change which properties are exposed to the user.   With a regular python document object, that logic can be implemented like the linear pattern.  If the user changes mode, some properties disappear and others appear.  This happens both in the panel and in the property pane by toggling the visibility of the properties.

You can't do that with just a shape file.  All the values in the varset/propertybag are visible all the time and the user just has to know which ones matter.

FEA-eng, 2026-09-20:
I assume that it's still relevant.