FreeCAD/FreeCAD #21433: FindHDF5 fails to find HDF5 after a failing pkg_search_module

xtexx, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In `cMake/FreeCAD_Helpers/SetupSalomeSMESH.cmake`, there are three paths of finding HDF5:
1. Indirectly, through VTK: https://github.com/FreeCAD/FreeCAD/blob/878f0b8c9c72c6f215833a99f2762bc3a3cf2abd/cMake/FreeCAD_Helpers/SetupSalomeSMESH.cmake#L58-L59
2. Directly, through pkg-config: https://github.com/FreeCAD/FreeCAD/blob/878f0b8c9c72c6f215833a99f2762bc3a3cf2abd/cMake/FreeCAD_Helpers/SetupSalomeSMESH.cmake#L98
3. Finally, through FindHDF5.cmake: https://github.com/FreeCAD/FreeCAD/blob/878f0b8c9c72c6f215833a99f2762bc3a3cf2abd/cMake/FreeCAD_Helpers/SetupSalomeSMESH.cmake#L99-L101

However, although VTK found HDF5 successfully, pkg-config fails and clears `HDF5_FOUND`, and FindHDF5 tries and also fails to find HDF5.

```
CMake Error at /usr/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find HDF5 (missing: HDF5_HL_LIBRARIES) (found version "1.14.6")
Call Stack (most recent call first):
  /usr/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.31/Modules/FindHDF5.cmake:1027 (find_package_handle_standard_args)
  CMakeLists.txt:1 (find_package)
```

This failure appears only when a sequence of `FindHDF5 pkg_search_module FindHDF5` is called. This is the minimal repro I found:
```cmake
find_package(HDF5 REQUIRED COMPONENTS C HL)
message("HDF5_FOUND: ${HDF5_FOUND}")
message("HDF5_LIBRARIES: ${HDF5_LIBRARIES}")
message("HDF5_C_LIBRARIES: ${HDF5_C_LIBRARIES}")
message("HDF5_HL_LIBRARIES: ${HDF5_HL_LIBRARIES}")

find_package(PkgConfig)
pkg_search_module(HDF5 hdf5-openmpi)
message("HDF5_FOUND: ${HDF5_FOUND}")
message("HDF5_LIBRARIES: ${HDF5_LIBRARIES}")
message("HDF5_C_LIBRARIES: ${HDF5_C_LIBRARIES}")
message("HDF5_HL_LIBRARIES: ${HDF5_HL_LIBRARIES}")

set(HDF5_PREFER_PARALLEL TRUE)
find_package(HDF5 REQUIRED)
message("HDF5_FOUND: ${HDF5_FOUND}")
message("HDF5_LIBRARIES: ${HDF5_LIBRARIES}")
message("HDF5_C_LIBRARIES: ${HDF5_C_LIBRARIES}")
message("HDF5_HL_LIBRARIES: ${HDF5_HL_LIBRARIES}")
```

Probably we should check `HDF5_FOUND` before calling pkg-config and reuse the result from VTK?

### Full version info

```shell
FreeCAD 1.0.1, CMake 3.31.6, HDF5 1.14.6, vtk 9.3.0

OS: AOSC OS (12.1.3) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.10.13, Qt 5.15.16, Coin 4.0.1, Vtk 9.3.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Project Tools & Websites

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-05-18:
In `main` this [commit](https://github.com/FreeCAD/FreeCAD/commit/9ea0f32692e13eee85b1e74bd42514942d357906) has been successful to the best of my knowledge, unfortunately it didn't get backported before the 1.0.1 release.

maxwxyz, 2025-05-19:
@adrianinsaval FYI

luzpaz, 2025-12-24:
> Yes, it is:

@xtexx are you building HEAD ?