FreeCAD/FreeCAD #21434: Translation of Sketch to Russian

VitaliyAT, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

So, I see 3 different names of Sketch in Russian translation:

1. _Sketch_ by itself in main menu
[image]

2. _Набросок_ in Properties
[image]

3. _Эскиз_ in Sketch menu
[image]

By my opinion it all should be **Эскиз** (3rd one)
Could you fix it in next reliese ?

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-05-18:
@VitaliyAT тема на форуме про переводы. Можете уточнить там.
https://forum.freecad.org/viewtopic.php?p=826996#p826996

maxwxyz, 2025-05-18:
Can be changed here https://crowdin.com/project/freecad

closed 2025-05-18 by maxwxyz