FreeCAD/FreeCAD #21435: Different icons in Constrains

VitaliyAT, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]
Constrains where point constrain to line looking different in different places.

[image]

[image]

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-18:
the combined tool is used for coincidence and point on object.

closed 2025-05-18 by maxwxyz

VitaliyAT, 2025-05-18:
Problem is that two different icons (look different) do the same things.