FreeCAD/FreeCAD #21436: BIM: Better organization of IFC Classification information in IFC models

semhustej, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**This is an idea for IFC objects:**

I think it might be appropriate to not place the Classification into the IFC group of parameters, but create IFC Classification group and place the parameter there.
I also think the parameter should be named Identification

The reasons for this is that Classification in IFC is not just a simple parameter. The steps above would make classification more standardized in FreeCAD. I am not concluding this based on my reading of IFC schema but looking how classification is exposed in BIMvision and Bonsai BIM tools, which I consider very good in regards to IFC standards.

Both of these have group for classification outside of Attributes and Properties. This is what lead me to the idea it might be good to have IFC Classification group in Freecad.
Both of these tool expose the classification code in parameter named Identification, so this leads me to the idea to name the parameter in FreeCAD the same way.

This is the proposition:

[image]

This is how classification looks like in BIM Vision:

[image]

This is how classification looks like in Bonsai BIM:

[image]

There is a lot more to IFC Classification (like the possibility to classify one object with multiple classification systems or separating Identification and Name), but this would create a good basis for later extending what is visible/possible in FreeCAD.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Anything else?

The proposal came from this related discussion: https://github.com/FreeCAD/FreeCAD/pull/21243

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tetektoza, 2025-05-18:
@semhustej Are you able to provide me the IFC file you've put in Bonsai BIM so it matched IFC Identification and Name from it?

semhustej, 2025-05-18:
This is the IFC file (remove the zip file extension): 

[IFC_sample_file_HVAC.ifc.zip](https://github.com/user-attachments/files/20272321/IFC_sample_file_HVAC.ifc.zip)

The exhaust part, that you can see in the screenshots above is Classified like this:

Classification System name: SNIM
Identification: VV.08
Name: Vyústka - Žaluzie

We will be using only `Identification` value as of now.

semhustej, 2025-05-18:
Note: If the classification value uses space, the first part is exported as `Classification System Name` and the second part as element `Identification`.

This currently does not work well in my opinion, but I would not change this at this point as I think in the future all this data should be separated in individual parameters in FreeCAD. But at this point it is not a priority and we can work with the simplified system.