FreeCAD/FreeCAD #21438: BIM: Add an algorithm for replacing/correcting windows, columns and beams via the taskbar.

kaiwas, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

At the moment, there is no convenient way to "replace" objects that have already been added to the document.
For example, to change a two-frame window to a single-frame one, we have to delete the existing window and install a new one.

At the moment, there is no convenient way to "replace" objects that have already been added to the document.
For example, to change a two-frame window to a single-frame one, we have to delete the existing window and install a new one.
A similar situation with a column.
In this case, it is almost impossible to save the original location of objects, simply because their location needs to be reconfigured
Video demonstration.

https://github.com/user-attachments/assets/52c6671e-9cf4-4be0-ac96-3a0d5e2876d2

We have an edit menu with the "edit" button, but I'm not sure that this can be used in any way. This is a completely incomprehensible function and I'm sure that no one uses it very often. This is some kind of complex and non-obvious function.

[image]

My suggestion. When you right-click in the "edit" view tree, a window should appear that appears when you first create an object. After that, you can change the "type" of the window and change its parameters.
Similar behavior is proposed for columns and beams.
[image]

Why? It's very simple - design is always a possibility of changes as development proceeds. Existing objects have to be changed no less often than new ones have to be created. But at the moment, despite the good parametrization capabilities, making changes is very difficult.
The video demonstrates how this is done in Archicad.
I also experimented with replacing thumbnails for windows in FreeCAD (configuration tables). The algorithm is crude, but you can still see how quickly and conveniently you can change a window.
I implemented a similar algorithm for msmenu profiles in this thread on the forum. Thanks to Roy043 for an important tip.
https://forum.freecad.org/viewtopic.php?t=88781

https://github.com/user-attachments/assets/d8715f04-2445-4afe-be1c-86bde5fb3e0c

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41726 (Git) AppImage
Build date: 2025/05/16 17:15:43
Build type: Release
Branch: main
Hash: 1ca108f8d503b2b68455844bfdfabb62fa1461ed
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master.zip
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * TrackingGroups
  * Curves 0.6.62
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-05-23:
@maxwxyz At least assign some kind of label ;)

maxwxyz, 2025-05-23:
IMO this should also work the same across WBs

kaiwas, 2025-05-23:
> IMO this should also work the same across WBs

Thank you! Yes, you are probably right.